Screenly/Anthias #346: Full screen images showing black bars - videos display full screen

pifiu, 2016-01-12:
All of the images I display appear to show black bars in the top and bottom, however the videos show in full screen without the bars. I tried the overscan setting but this does not resolve it.

I know the recommended size is 1920x1080, but ocassionally I use 1620x1080 and both display black bars at the top and bottom.

vpetersson, 2016-01-12:
If you're using images any smaller than 1920x1080, you will get a border. That's expected, as we never scale up images. As such it's expected that you get a black border with 1620x1080. 

If you are still getting a black border with images that are 1920x1080, it is almost certainly an overscan issue.

closed 2016-01-12 by vpetersson

pifiu, 2016-01-12:
I tried an image of 1920x1280 and it still shows the black bars. 

Also I changed the background color from black to the color that matches the images I am using.

pifiu, 2016-01-22:
A couple of notes.

I changed the background color of the document from black to a color that matches our images (yellow) and here is the result of more testing:

1920x1080 images - Black bars on the top and bottom, and on the side I get the yellow background. 
1920x1280 - Black bars on the top and bottom, the sides fill in completely.

The TV is a Samsung RM48D running at 1920x1080 / 60p. 

The videos run full screen with no bars, so overscan is not the issue. I even tried toggling it on and off and it was worse since it actually stretched the image past the screen. If it was an overscan issue I would have a black boder around the videos and I do not.

Any suggestions?

I can provide pictures if required.

vpetersson, 2016-01-22:
> The videos run full screen with no bars, so overscan is not the issue.

This is an incorrect conclusion. The video player talks directly to the frame buffer. This is different from X11. Hence, it's most likely an overscan issue.

pifiu, 2016-01-22:
If I toggle overscan, the videos display larger than the screen and so do the images. What do you propose I do to troubleshoot?

vpetersson, 2016-01-22:
In that case, you'd have to play with the `overscan_*`  values. Please note that these can be both positive and negative. For more information, please see:
https://www.raspberrypi.org/documentation/configuration/config-txt.md

pifiu, 2016-01-22:
Is there a reason why one should have to play with the settings? Why is it that on some displays it just works? What causes that?

If I alter the overscan settings wont that affect the videos as well?

vpetersson, 2016-01-22:
> Is there a reason why one should have to play with the settings?
> Why is it that on some displays it just works? What causes that?

Generally speaking, most modern TVs works out-of-the-box. 

This issue however doesn't have anything to do with Screenly. It's something that affects all Raspberry Pis.

> If I alter the overscan settings wont that affect the videos as well?

No, it shouldn't.

zer0ring, 2016-01-23:
Chris Ryan, Lord of Blackwood

Date: Fri, 22 Jan 2016 08:52:03 -0800