Screenly/Anthias #352: Add option to backup

pifiu, 2016-01-22:
I run the raspberry pi 24/7 behind a TV using Screenly OSE and ocassionally when there is a power surge, the file system seems to get corrupt and its very time consuming to reinstall, and upload all of the content again.

Is there a way to generate a backup of the config with optionally the content as well? That way if someone just wants to save the configuration (which there is really little to nothing to configure) by itself they can do that, or if they want the configuration + the content they can as well.

It seems like it would make reloading things easier in the case the file system gets corrupt.

vpetersson, 2016-01-22:
It's an interesting idea. However, for the time being, my recommendation would be to simply backup the SD card directly. If you're on Linux or OS X, you can do that easily using `dd` for instance.

vpetersson, 2016-04-26:
Maybe we should add this functionality. All we really need to do is to create an archive of: 
* `~/.screenly`
* `~/screenly_assets`

There could however be issues with creating these if there are lots of large assets on disk, so we probably need to have this performed async by some background daemon.

krzimmer, 2016-09-23:
If I wanted to move all assets from one screenly pi to another, all I would need to do is copy
~/.screenly
~/screenly_assets
to the other device?

vpetersson, 2016-09-24:
Yup. That should be sufficient, assuming Screenly is properly installed.

krzimmer, 2016-09-29:
I tried doing this.  After the transfer, Screenly will not launch on the target device.  I used the Screenly Image and verified that the target device was working properly prior to the transfer.  Any suggestions?

vpetersson, 2016-09-30:
Hard to say without any logs.

pifiu, 2016-11-02:
so is the /screen_assets/ the only thing needed or is ~/.screenly also required for a backup?

vpetersson, 2016-11-03:
You need to backup `~/.screenly` as well, as that is where the database and config resides.

vpetersson, 2016-12-21:
@over64 hold off with this one a bit as we need to think of a good way to store this data such that we can later restore it easily. Right now I'm thinking BSON would be a good format for this.

vpetersson, 2017-06-12:
Closed in #563.

closed 2017-06-12 by vpetersson