Screenly/Anthias #354: Create new disk image based on Jessie Lite

vpetersson, 2016-01-26:
With the release of Jessie, there's now a [Lite version](https://www.raspberrypi.org/downloads/raspbian/) available. It makes a lot of sense for Screenly to use this as it would create a much smaller footprint.

    $ du -hs 2015-11-21-raspbian-jessie-lite.zip
    375M    2015-11-21-raspbian-jessie-lite.zip

over64, 2016-02-04:
I have tried jessie lite just now. Bare system has 900Mb size. Lxde installation requires 800Mb additional space. 1,7Gb overall. Is it better then full distro?

vpetersson, 2016-02-04:
I'm not sure if we even need LXDE. We might as well just load `xloader.sh` directly as a WM.

vpetersson, 2016-02-04:
Actually, can't we just make `xloader.sh` into `.xinitrc` instead?

pearson-lucas-dev, 2016-02-12:
I was actually looking at doing this exact thing has anyone started yet?  I have done something similar using minibian: https://drive.google.com/a/five-startech.com/file/d/0ByyG_hCIHsenZGdIWGhscXhQRkk/view?usp=sharing

I would love to move that to screenly (screenly is much cleaner than what I did) but it should still stay ~ 1GB not sure if that is worth the "unoffical" title that minibian might be perceived as verse jessie lite

Just some thoughts love the work you've all done!

vpetersson, 2016-02-12:
@pearson-lucas-dev Thanks for the kind words. I've requested access to the file. 

Also, yes, we have begun work on this. We would prefer to stick with one of the official images, and Jesse Lite is pretty slim (but there's clearly room for improvement).

@over64 is working on this and we hopefully have something in the near future.

Captain-DL, 2016-03-22:
Has there been any progress with this? We just got 6 shiny new Pi 3s and can't load Wheezy on them. :( Hopefully this is close because we got them specifically for Screenly OSE.

vpetersson, 2016-03-22:
Yes, a fair amount of progress, but we're refactoring the setup to make it easier to maintain as well as to use systemd. Here's the PR #368.

Captain-DL, 2016-03-24:
Not sure if you have made changes to the distro, but I did a manual load this morning on Jessie (not lite) and it is working perfectly!

vpetersson, 2016-04-26:
Done. Latest images are based on Jessie Lite (and the installation works).

closed 2016-04-26 by vpetersson