Screenly/Anthias #357: Video Playback Delayed

bash, 2016-02-15:
Hey there

We are using a Screenly version from around July 2015 and recently came across an issue with the video playback:
When a video starts the first frame is displayed for about 20 seconds until the video starts playing.

I should maybe add that we didn't have any issues when we initially set up our Raspberry PI a few month back.

Are there any known issues/solutions to this?

vpetersson, 2016-02-15:
Very strange. I don't think it's Screenly related per see,. My guess is that it's the SD card that's about to give up. Try replacing it.

bash, 2016-02-16:
Thank you for your quick answer. We'll try using a new SD card.
I'll post back once we've tried it out.

Kentzo, 2016-02-25:
@bash A bit of advice while looking for a new SD card. Good vendors specify type of the memory used somewhere. If they don't I'd avoid them. If they do, prefer SLC over MLC over TLC in terms of durability.

bash, 2016-03-23:
Quick Update: I have tried with 2 diffrent SD cards and it brings only a small improvement.
The problem seems to be omxplayer itself and I have experimented with diffrent flags without any success.

Mainly I have 2 questions:
- Is this a known problem with GoPro videos?
- Is there a possibility in Screenly Pro to show an image while the video is loading instead of having a black screen?

vpetersson, 2016-03-23:
@bash 

> Is this a known problem with GoPro videos?

I'm not sure what file format GoPro uses, but the Raspberry Pi only supports H264 videos out-of-the-box. As such, they'd need to be re-encoded to this if you're using OSE (in Pro we do that automatically).

> Is there a possibility in Screenly Pro to show an image while the video is loading
> instead of having a black screen?

If you have a Class 10 SD card, there shouldn't be much of a delay.  We are however actively working on reducing/eliminating this in Screenly Pro.

vpetersson, 2016-04-26:
Closing out as we're able to reproduce this.

closed 2016-04-26 by vpetersson