Screenly/Anthias #358: glibc6 (CVE-2015-7547 / DSA-3481-1)

scrns, 2016-02-17:
Hi, I noticed that http://apt.screenlyapp.com/raspbian/ (defined in `/etc/apt/sources.list` with the disk image download) doesn't seem to have the latest libc6 packages ([2.13-38+deb7u10](https://security-tracker.debian.org/tracker/CVE-2015-7547)) which has a fix for the recent glibc dns client resolver RCE bug. Could the repository be updated?

vpetersson, 2016-02-17:
Yessir. Good catch. Update should be live in the next few hours.

closed 2016-02-17 by vpetersson

cegichitho, 2016-02-23:
Hi, has the repo been updated? I don't think I have the fixed libc6 (showing `2.13-38+rpi2+deb7u8` as latest on the pi)

cegichitho, 2016-02-24:
ping @vpetersson

cegichitho, 2016-02-25:
Is there any difference between screenly's repos and raspian's?

vpetersson, 2016-02-26:
> Is there any difference between screenly's repos and raspian's?

Screenly is just a 'locked' mirror of the official Raspbian apt repository. 

I didn't check the exact version that got in in the last sync, but we did update the mirror when I posted that before.

I will initiate a new mirror refresh shortly.