Screenly/Anthias #359: PI hanging and rebooting after update

curtis45, 2016-02-18:
My pi 2B has been running great the last 6 months, then I decided to run an update 3 days ago and it hasn't been running right since.
It locks up and the screen goes black for roughly 5 min. randomly, then the pi restarts and goes for hours and then will lock up again.
[image]

vpetersson, 2016-02-18:
Try re-flashing the SD card with the disk image and run the upgrade again.

curtis45, 2016-02-18:
wow, fast response, before I even had a chance to upload screen shots

curtis45, 2016-02-20:
OK, formatted card, re-flashed with latested download of screenly, adjusted settings in raspi-config, updated screenly and loaded assets. ran for about an hour and completely locked up with no auto reboot, so I have no screen shot of logs.
Current running assets are 3 small videos and a webpage to here: http://www.wunderground.com/q/zmw:83530.1.99999 for weather.

curtis45, 2016-02-20:
Webpage seems to be the problem, removed it and so far so good, would really like to get that back though.

curtis45, 2016-02-20:
Well its not the webpage, locked up again about an hour after turning it off. :(

curtis45, 2016-02-22:
OK,,,, I left only 1 mp4 video on over the weekend, it is 1 that I made from PowerPoint 2013. Worked perfectly all weekend. So my problem has to be another video I have been using, the only problem with that is that I have no new videos, so the update has a problem with some sort of video settings. I can post media info images if you want.

curtis45, 2016-02-22:
Well it crashed again, but did restart on its own, this was looping only my video and the weather webpage.
[image]

demethan, 2016-03-07:
To dev, might be related. Check issue #361  Thx

curtis45, 2016-03-07:
Just to follow up demethan, I re-installed Screenly again and did not do the update. I put on everything I had before including the webpage and I am having no problems. There is definitely something wrong with the update.

demethan, 2016-03-08:
I did the same. Flash the mSD card again with the img and everthing is fine if you don`t update.

closed 2016-04-25 by vpetersson