Screenly/Anthias #360: Webpages do not refresh after the initial loading

xdozex, 2016-02-23:
I'm using Screenly to display a few different websites that all have dynamic stats. Each website has content that will update/change every 5 - 10 minutes and it's crucial that the viewers have access to the most up-to-date information. 

I'm able to get the website to load up initially and the website displays the most up-to-date content, but then once the playlist cycles back to the same page again, it shows the information that was loaded previously. Meaning the website never actually appears to refresh. 

Is there an easy way to force Screenly to refresh any/all websites when it reaches them in the playlist cycle?

enieuwenhuis, 2016-03-26:
same problem here. I'd like to know the solution

vpetersson, 2016-04-28:
Hmm, that's strange. Could it be server side caching? 

Could you provide steps so that we can reproduce it (with real page)?

xdozex, 2016-04-28:
To be honest, I'm not really doing anything out of the ordinary with it. The pages I was trying to load were hosted on Digital Ocean and are quite literally, an extremely minimal HTML page with a few divs, headers, 2 paragraph sections and one image. I made 3 copies of the page and changed the text slightly between images so I could load in 3 "sites" and have them rotate. Everything appeared to be working properly until I went to make a change to the copy on one of the pages. 

I edited one of the headers on the page and saved it up to the server. But when Screenly got back to the site in the rotation, the previous version of the page was still loading. The only way I could get the page to refresh was to shut down Screenly and reboot the Pi. Once it was back up it would load the correct, new version of the page, but once again if any changes happen after that page was loaded initially, it would only use the first version it loaded in. Made it impossible to continue using Screenly for our needs, so it would be nice if this could be sorted. We'd love to use it for our in-house display. 

It really seems like when it loads it just needs to force a page refresh quickly as it's about to be displayed in the rotation.

xdozex, 2016-04-28:
Sorry, forgot to include a link to the page that was being loaded: http://canalalarm.co/signage/product1.html

vpetersson, 2016-05-03:
This is a duplicate/derivation of #315.

closed 2016-05-03 by vpetersson