Screenly/Anthias #361: Can not get to 8080 page after update

CrazyBoySnp, 2016-02-25:
I just installed a new image of screenly onto my raspberry Pi.

After did the update and rebooted and no my screen display shows up BUT I can not get to the back end OSE to add and change pages as I get a This webpage is not available

ERR_CONNECTION_REFUSED

This was work Before doing the Screenly Upgrade as requested by the software.

Wondering what can be done to fix this?

demethan, 2016-03-07:
Hi, I have the same problem as @CrazyBoySnp.

Screenly OSE console asked for update. I ran ~/screenly/misc/run_upgrade.sh from ssh.

After the reboot, the back end console refused connection through http:/xx.xx.xx.xx:8080

vpetersson, 2016-04-26:
Try running the upgrade, then reboot, and run the upgrade again.

closed 2016-04-26 by vpetersson

nateATcanarchy, 2022-11-18:
im having the same issue - the MISC directory on my installation does not exist.  is there a different path for a RASPI 3+b ?

vpetersson, 2022-11-19:
@nateATcanarchy try doing a fresh reinstall on Bullseye.