Screenly/Anthias #367: Server Error: Could not retrieve file. Check the asset URL.

efinst0rm, 2016-03-16:
Hello,

I've been having this issue with one of my locations while I have two other pi's running without any issues. With the same link, when I do curl -I it returns 

`HTTP/1.1 200 OK
Server: nginx
Date: Wed, 16 Mar 2016 22:24:08 GMT
Content-Type: text/html; charset=UTF-8
Connection: keep-alive
Keep-Alive: timeout=60`

Thank You.

vpetersson, 2016-03-17:
It could be something related to the user agent. Try the following from the Raspberry Pi:

```
$ python
> import requests
> r = requests.head('http://yoururl.com')
> print r.status_code
> print r
```

efinst0rm, 2016-03-23:
Thanks for the reply this is the error I get when i run r = requests.head('myurl')
`Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/usr/local/lib/python2.7/dist-packages/requests/api.py", line 95, in head
    return request('head', url, **kwargs)
  File "/usr/local/lib/python2.7/dist-packages/requests/api.py", line 50, in request
    response = session.request(method=method, url=url, **kwargs)
  File "/usr/local/lib/python2.7/dist-packages/requests/sessions.py", line 465, in request
    resp = self.send(prep, **send_kwargs)
  File "/usr/local/lib/python2.7/dist-packages/requests/sessions.py", line 573, in send
    r = adapter.send(request, **kwargs)
  File "/usr/local/lib/python2.7/dist-packages/requests/adapters.py", line 415, in send
    raise ConnectionError(err, request=request)
requests.exceptions.ConnectionError: ('Connection aborted.', gaierror(-5, 'No address associated with hostname'))
`

vpetersson, 2016-03-24:
So I don't understand how you can 'No address associated with hostname' from requests but a '200 OK' from `curl`. That doesn't make sense.

It could be a DNS issue perhaps if you're running them on different computers.

vpetersson, 2016-04-28:
Closing out as we're unable to reproduce.

closed 2016-04-28 by vpetersson

efinst0rm, 2016-04-30:
Sorry for the horribly late responds but issue was on my end with the DNS. Thank for the help!