Screenly/Anthias #369: Unable to show google calendar in Screenly - hosted via Dropbox

luedm, 2016-03-18:
after following this guide: https://www.screenlyapp.com/use-cases/dashboard/google-calendar/

I am unable to correctly display my calendar. I followed the instructions and included the embed url in a notepad file, changed the extension to .html, and uploaded and shared the link from Dropbox. 

When I share the link to the file, it simply shows the file, without launching it.

Does dropbox not allow these permissions anymore?

vpetersson, 2016-03-18:
Yeah, we need to update that. Dropbox changed this recently and you can no longer serve HTML from there.

luedm, 2016-03-18:
Does any other work around exist?

vpetersson, 2016-03-18:
Sure, any web hosting service should do.

vpetersson, 2016-04-26:
Try running the upgrade. This should hopefully have been fixed now.

closed 2016-04-26 by vpetersson