Screenly/Anthias #370: How to display a website with cookies

enieuwenhuis, 2016-03-29:
Hi, I 'd like to display, for example,  the website http://www.buienradar.nl/, but when I display this I get a cookie notification. 
How can I clear that notification ?
[image]

Best regards
Erik

vpetersson, 2016-03-29:
I can't think of any (easy) way to clear that. The only possible way that I can think of would be to connect a mouse/keyboard, accept the pop-up and then disconnect them

enieuwenhuis, 2016-03-29:
Thanks for your reply. 
I tried this, but the mouse does not work in screenly,

vpetersson, 2016-03-29:
It should work, but you might not see the cursor.

forum6691, 2016-11-04:
Is it a solution, or a way to validate this cookie outside screenly ? I tried to switch in admin mode, logging and type "startx" in order to access to browser, but the command is inefective.

vpetersson, 2016-11-04:
@forum6691 no you can't `startx` as X is already running. You might however be able to connect a mouse.