Screenly/Anthias #372: SCREENLY || ImportError / doesn't start

smartye, 2016-04-03:
Hi,

my screenly doesn't start. I looked at "screenly_xloader.log" and there is this message in loop:

_root@screenly:/tmp# tail -f screenly_xloader.log
Traceback (most recent call last):
  File "/home/pi/screenly/viewer.py", line 8, in <module>
    from datetime import datetime, timedelta
ImportError: /usr/lib/python2.7/lib-dynload/datetime.so: undefined symbol: PyExc_ValueError
Cleaning up...
Traceback (most recent call last):
  File "/home/pi/screenly/viewer.py", line 8, in <module>
    from datetime import datetime, timedelta
ImportError: /usr/lib/python2.7/lib-dynload/datetime.so: undefined symbol: PyExc_ValueError
Cleaning up..._

Can anyone give me some help? Thanks in advance

vpetersson, 2016-04-03:
Seems like something is broken in Python. I'd re-flashing the SD card.

closed 2016-04-03 by vpetersson

smartye, 2016-04-04:
Ok, thank you.
please close this issue.