Screenly/Anthias #375: Updating

rsenio, 2016-04-19:
Unable to update either one of my screenly installs. Noticed today there was an update available. As the 'pi' user, running either 

 ~/screenly/misc/run_upgrade.sh
or
curl -sL https://raw.github.com/wireload/screenly-ose/master/misc/upgrade.sh | bash

results in bash: line 1: Not: command not found

I update these when there are updates available, perhaps this stopped working the last update

rsenio, 2016-04-19:
Apparently the command should be 
~/screenly/misc/upgrade.sh, not what it stated on the website

jscaptura, 2016-04-19:
I can confirm the same result on my install.

rsenio, 2016-04-19:
Now the web gui isn't working after installing the update today, and nothing is displayed on the screen (black screen). 

This is just great

vpetersson, 2016-04-19:
There are a fair amount of changes in this update behind the scenes. Let me see if I can reproduce this.

In the meantime, please run:

```
curl -sL https://raw.github.com/wireload/screenly-ose/master/misc/install.sh | bash
```

rsenio, 2016-04-19:
This is the end result of running 
curl -sL https://raw.github.com/wireload/screenly-ose/master/misc/install.sh | bash

TASK [Enable screenly systemd services] ****************************************
failed: [localhost] => (item=X.service) => {"changed": true, "cmd": ["systemctl", "enable", "X.service"], "delta":                                                                                                                            "0:00:00.140307", "end": "2016-04-19 09:20:14.245042", "failed": true, "item": "X.service", "rc": 1, "start": "20                                                                                                                           16-04-19 09:20:14.104735", "stderr": "ln -s '/etc/systemd/system/X.service' '/etc/systemd/system/multi-user.target                                                                                                                           .wants/X.service'", "stdout": "", "stdout_lines": [], "warnings": []}
failed: [localhost] => (item=matchbox.service) => {"changed": true, "cmd": ["systemctl", "enable", "matchbox.servi                                                                                                                           ce"], "delta": "0:00:00.139318", "end": "2016-04-19 09:20:18.844143", "failed": true, "item": "matchbox.service",                                                                                                                            "rc": 1, "start": "2016-04-19 09:20:18.704825", "stderr": "ln -s '/etc/systemd/system/matchbox.service' '/etc/syst                                                                                                                           emd/system/multi-user.target.wants/matchbox.service'", "stdout": "", "stdout_lines": [], "warnings": []}
failed: [localhost] => (item=screenly-viewer.service) => {"changed": true, "cmd": ["systemctl", "enable", "screenl                                                                                                                           y-viewer.service"], "delta": "0:00:00.123803", "end": "2016-04-19 09:20:23.332658", "failed": true, "item": "scree                                                                                                                           nly-viewer.service", "rc": 1, "start": "2016-04-19 09:20:23.208855", "stderr": "ln -s '/etc/systemd/system/screenl                                                                                                                           y-viewer.service' '/etc/systemd/system/multi-user.target.wants/screenly-viewer.service'", "stdout": "", "stdout_li                                                                                                                           nes": [], "warnings": []}
failed: [localhost] => (item=screenly-web.service) => {"changed": true, "cmd": ["systemctl", "enable", "screenly-w                                                                                                                           eb.service"], "delta": "0:00:00.127936", "end": "2016-04-19 09:20:27.855465", "failed": true, "item": "screenly-we                                                                                                                           b.service", "rc": 1, "start": "2016-04-19 09:20:27.727529", "stderr": "ln -s '/etc/systemd/system/screenly-web.ser                                                                                                                           vice' '/etc/systemd/system/multi-user.target.wants/screenly-web.service'", "stdout": "", "stdout_lines": [], "warn                                                                                                                           ings": []}

NO MORE HOSTS LEFT *************************************************************
        to retry, use: --limit @screenly.retry

PLAY RECAP *********************************************************************
localhost                  : ok=8    changed=2    unreachable=0    failed=1

vpetersson, 2016-04-19:
Yes, i was able to reproduce it on my end too. Working on it.

over64, 2016-04-19:
@rsenio, it's should work after this stack trace. Just need to reboot

jscaptura, 2016-04-19:
Running:

curl -sL https://raw.github.com/wireload/screenly-ose/master/misc/install.sh | bash

Twice resulted in a successful install.

vpetersson, 2016-04-19:
Yeah, the root cause is that we're switching from `supervisord` to `systemd` (which is native in Jessie). Apparently installing `systemd` requires a reboot. 

Hence, if someone else stumbles across this, the solution is:

    curl -sL https://raw.github.com/wireload/screenly-ose/master/misc/install.sh | bash

Issue a reboot (`sudo shutdown -r now`)

Run the installation again (may not be necessary):

    curl -sL https://raw.github.com/wireload/screenly-ose/master/misc/install.sh | bash

closed 2016-04-19 by vpetersson

over64, 2016-04-19:
@rsenio there is no need twice) It should be:
1) run install.sh
2) maybe got ansible trace on system which booted not on systemd
3) reboot
4) should work

rsenio, 2016-04-19:
Both screenly installs are showing assets now for me, after running 
curl -sL https://raw.github.com/wireload/screenly-ose/master/misc/install.sh | bash

rsenio, 2016-04-21:
So, today both installs that were "repaired" by running "curl -sL https://raw.github.com/wireload/screenly-ose/master/misc/install.sh | bash" show an update available. Is this is same update that broke everything last time? Is it safe to update?

vpetersson, 2016-04-21:
It's safe to update. You can check the commit log for details on what changed.

thebongmaster, 2016-04-25:
Having trouble with the update. I ran the update and i get:

`TASK [Copy gtk config] *********************************************************
fatal: [localhost]: FAILED! => {"changed": false, "checksum": "b05a6d76bb4cba53d4e4d660e7642136ab753413", "failed": true, "gid": 1000, "group": "pi", "mode": "0777", "msg": "src file does not exist, use \"force=yes\" if you really want to create the link: /home/pi/gtkrc-2.0", "owner": "pi", "path": "/home/pi/.gtkrc-2.0", "size": 32, "src": "gtkrc-2.0", "state": "link", "uid": 1000}

NO MORE HOSTS LEFT *************************************************************
        to retry, use: --limit @screenly.retry

PLAY RECAP *********************************************************************
localhost                  : ok=3    changed=0    unreachable=0    failed=1
`
I have tried reboting several times and re-running the `$ curl -sL https://raw.github.com/wireload/screenly-ose/master/misc/install.sh | bash ` command and i get the same thing. 
now all i get is a login prompt on the screen. please help, we need this working

vpetersson, 2016-04-25:
Hmm. Strange. I don't see anything obvious. Try a manual `git pull` perhaps?

Also, we're making a number of improvements in #380 so you might want to check out that branch.

thebongmaster, 2016-04-25:
thanks, not to sound dumb, but how do i do a git pull? (im not used to github)

vpetersson, 2016-04-25:
```
cd ~/screenly
git reset --hard
git pull
```
(The `git reset` is in case there are any dirty changes.)

thebongmaster, 2016-04-25:
tried that, got this:
`pi@infoscreen ~/screenly $ git reset --hard
HEAD is now at 3c34541 Merge pull request #366 from robberwick/fix_24_hr_date_format
pi@infoscreen ~/screenly $ git pull
Already up-to-date.
`

vpetersson, 2016-04-25:
I was able to reproduce this on a very old image. I've pushed an update. Could you try re-running the `curl` command?

vpetersson, 2016-04-25:
If you're upgrading from an old version of Screenly, you may also need to run:
```
sudo apt-get install systemd-sysv
```

thebongmaster, 2016-04-25:
i tried a fresh install (4/20) will run the update after i finish restoring the assets and will let you know

thebongmaster, 2016-04-25:
All was fine until i decided to run the command to enable SSL. now the web interface looks a little broken and the 'add asset' button does nothing

vpetersson, 2016-04-25:
@thebongmaster You probably got stuck in some limbo state between the old and new version.

thebongmaster, 2016-04-26:
since the update none of my vids seem to be working, one plays 1st time round and upon the next loop of the playlist it goes to a black screen and stays that way from there on out. the other just flashed colours on the screen. have had to disable them til i can work out why. the vids played perfectly before the updates