Screenly/Anthias #382: Stuck in the black screen

CrownWine, 2016-04-21:
Hi.

I configure a screenly with the latest version 04-18-2016 on a Raspberry pi 3, and all looks fine, but it keep stuck in the black screen after a few assets cycle shows, it is anyway to avoid the screenly to keep in the black page.

Regards.

[image]

CrownWine, 2016-04-21:
I just take this new image (I left the raspberry on for a long time) watch the elapsed time and the few assets that went well

[image]

vpetersson, 2016-05-03:
@CrownWine Could you run the following commands:

```
$ bash <(curl -sL https://www.screenlyapp.com/install-ose.sh)
$ export PYTHONPATH="$HOME/screenly"
$ python ~/screenly/lib/diagnostics.py
```

Then post the output of the diagnostics command here.

dsency, 2016-05-05:
Hello,
I am having the same problem.  The screen goes blank after a seemingly random amount of time.  A reboot always returns things to normal.  Also, after doing an update yesterday, I am no longer getting output in the Viewer Log.  I've attached files containing the output of the bash upgrade command, diagnostics command and a screenshot of the Viewer Log message.

[diagnostics.txt](https://github.com/wireload/screenly-ose/files/250450/diagnostics.txt)
[image]
[upgrade.txt](https://github.com/wireload/screenly-ose/files/250451/upgrade.txt)

vpetersson, 2016-05-06:
I've looked over the attached data but can't find anything obvious. 

Could you try fetching the logs by running:

```
$ journalctl -u screenly\*
```

dsency, 2016-05-06:
output of journalctl attached.
[journalctl-output.txt](https://github.com/wireload/screenly-ose/files/252206/journalctl-output.txt)

dsency, 2016-05-06:
the journal contents stopped updating at the May 06 08:34:48 date/time stamp and screenly has gone black.

vpetersson, 2016-05-06:
Try `systemctl restart screenly*` and then check the logs.

dsency, 2016-05-06:
got an error trying to run that command:

pi@raspberrypi:~ $ sudo systemctl restart screenly*
Failed to restart screenly.service: Unit screenly.service failed to load: No such file or directory.
Failed to restart screenly_assets.service: Unit screenly_assets.service failed to load: No such file or directory.
pi@raspberrypi:~ $

vpetersson, 2016-05-06:
Ah, you might need to escape that `*` by with:

    $ sudo systemctl restart screenly\*

dsency, 2016-05-06:
yup, adding the backslash made the viewer restart but it only showed the first item in my list before going back to a black screen.  i've attached a file containing a the current output of journalctl -u screenly\*
do you need anything else?
[journalctl-output-2.txt](https://github.com/wireload/screenly-ose/files/252677/journalctl-output-2.txt)

dsency, 2016-05-06:
also the viewer log is still blank and no log file is being created in the /tmp folder.

vpetersson, 2016-05-06:
Looks like it's truncated. Could you redirect the output to a file instead?

    $ journalctl -u screenly\* > ~/logs.txt

> also the viewer log is still blank and no log file is being created in the /tmp folder.

Yup. That's expected as we've moved to systemd for logging.

dsency, 2016-05-06:
the truncated area is the time the screen is black and the
[logs.txt](https://github.com/wireload/screenly-ose/files/252707/logs.txt)
 viewer is not displaying anything.

vpetersson, 2016-05-09:
I can see a strange timeout issue in the logs. Are you running anything else on the Raspberry Pi?

dsency, 2016-05-09:
nope.  standard, off-the-shelf raspberry pi 3 model b v1.2 and screenly ose 2016-04-20 with applicable screenly updates.

vpetersson, 2016-05-09:
What kind of SD card?

dsency, 2016-05-09:
sandisk ultra 8gb class 10

dsency, 2016-05-11:
since it appeared we had hit a troubleshooting wall i decided to try a new approach.  i loaded an identical sd card with the 2016-04-20-Screenly_OSE_4GB image into the pi and let it run the default content.  eventually, this setup exhibited the blank screen behavior.  ran the following commands and the diagnostic outputs are attached:

`$ bash <(curl -sL https://www.screenlyapp.com/install-ose.sh)`
`$ export PYTHONPATH="$HOME/screenly"`
`$ python ~/screenly/lib/diagnostics.py`
`$ journalctl -u screenly\*`

[diagnostics.py_5-11-2016.txt](https://github.com/wireload/screenly-ose/files/259299/diagnostics.py_5-11-2016.txt)

[journalctl_5-11-2016.txt](https://github.com/wireload/screenly-ose/files/259307/journalctl_5-11-2016.txt)

vpetersson, 2016-05-11:
@dsency The logs are truncated from the journaling again :)

There's something in there though. Could you try doing a reboot to make sure everything was applied properly and then try to reproduce it?

dsency, 2016-05-11:
Okey dokey.  Stay tuned...

dsency, 2016-05-12:
ok, appearance of the blank screen of death was much quicker after the updates and a reboot.  log files attached.
[diagnostics_5-12-2016.txt](https://github.com/wireload/screenly-ose/files/261432/diagnostics_5-12-2016.txt)
[logs_5-12-2016.txt](https://github.com/wireload/screenly-ose/files/261431/logs_5-12-2016.txt)

dsency, 2016-05-17:
tried another approach.  i loaded a fresh sd card of the same type with the 2016-04-20-Screenly_OSE_4GB image into a pi 2 model b+, ran the update command (had to do it twice) and let it run the default content.  after around 20 hours of continuous use the blank screen of death appeared.  log files attached.
[diagnostics_5-17-2016.txt](https://github.com/wireload/screenly-ose/files/268476/diagnostics_5-17-2016.txt)
[journalctl_5-17-2016.txt](https://github.com/wireload/screenly-ose/files/268477/journalctl_5-17-2016.txt)

richardplenty, 2016-05-18:
I am having problems with the new version on a pi3 sandisk ultra 128g. My videos do not show up, just black, but the cartoon you supply does play but then the system stops.  My assets run fine (same files) on my Pi 2 previous screenly version. info: Load Average Free Space Uptime Monitor Info 0.09 110G 0 days and 0 hours 0x12000a [hdmi cea (16) rgb lim 16:9] 1920x1080 @ 60.00hz Progressive 

LATEST VIEWER LOGS Tue, 17 May 2016 23:17:16 Loading browser... Tue, 17 May 2016 23:17:16 Browser loading file:///tmp/screenly_html/black_page.html. Running as PID 735. Tue, 17 May 2016 23:17:19 Generating asset-list... Tue, 17 May 2016 23:17:19 Showing asset Big Buck Bunny (video) Tue, 17 May 2016 23:29:23 current url is file:///tmp/screenly_html/black_page.html Tue, 17 May 2016 23:29:58 Showing asset IMG00021-20100626-2126.jpg (image) Tue, 17 May 2016 23:29:58 Sleeping for 10 Tue, 17 May 2016 23:30:08 Showing asset JohnnyJordaanplein41U.mp4 (video) Tue, 17 May 2016 23:30:09 current url is file:///tmp/screenly_html/black_page.html

vpetersson, 2016-05-20:
I just tried to reproduce this, but was unable to do so. 

@dsency According to the logs, you have no active assets. That would explain the black screen.
```
May 16 17:32:31 raspberrypi python[584]: Playlist is empty. Sleeping for 5 seconds
May 16 17:32:36 raspberrypi python[584]: Playlist is empty. Sleeping for 5 seconds
May 16 17:32:41 raspberrypi python[584]: Playlist is empty. Sleeping for 5 seconds
May 16 17:32:46 raspberrypi python[584]: Playlist is empty. Sleeping for 5 seconds
May 16 17:32:51 raspberrypi python[584]: Playlist is empty. Sleeping for 5 seconds
May 16 17:32:56 raspberrypi python[584]: Playlist is empty. Sleeping for 5 seconds
```

@richardplenty You're running on an old version of Screenly. Please upgrade. The latest version works fine on the Raspberry Pi 3. I have one sitting on my desk right here. Make sure you run:
`bash <(curl -sL https://www.screenlyapp.com/install-ose.sh)`

closed 2016-05-20 by vpetersson

dsency, 2016-05-20:
@vpeterson I don't agree that this bug report should be closed.  It may be that my experiment with the older model pi with the latest version of screenly may have aged out the default asset list since i did no post installation configuration like setting the time zone, but my original problem report was for a pi 3 with the latest version of screenly and a playlist with active assets.  you even commented that you saw a strange timeout issue in the logs from that setup.  am i to assume that i need to retest for the blank screen behavior and file a duplicate bug report if i see still the problem?

vpetersson, 2016-05-20:
I'm going to try to reproduce this on a Pi Zero as I don't have any older Pis laying around here.

dsency, 2016-05-20:
will using a pi zero be a good test since it's not the same hardware i'm using?  i might be willing to part with a pi 3 to help with troubleshooting.

vpetersson, 2016-05-21:
@dsency I have plenty of Pi 2 and Pi 3s but I have not been able to reproduce it on them. 

If you're able to provide exact steps on how to reproduce it from start to finish, we will likely be able to fix it.

richardplenty, 2016-05-22:
I did do the update requested    What I have found is that many of my videos that played on pi2 will no longer play on the pi3 using the 4-20 version.  All are 16mb files.  My 12 my files seem to work.  Duration does not seem to be an issue.   Also my logging is blank but there is a line saying "no viewer log present -is only the screenly server running?"  Finally whenever there is a video transition or I am loading a video I see a small box in the upper right portion of the frame with colors inside. Not sure why that appears.

vpetersson, 2016-05-23:
> no viewer log present -is only the screenly server running?

That's expected. A fix is in the work.

> Finally whenever there is a video transition or I am loading a video I see a small box in the upper right portion of the frame with colors inside.

If you're getting a rainbow in the upper right-hand corner, that's a sign that the device isn't getting enough power. The Raspberry Pi 3 does consume more power than the Raspberry Pi 2, so it's not unlikely that the power supply is the issue.

mualemi, 2016-07-05:
I might be jumping the gun here...
I had the same issue with the black screen on a clean install of the  2016-04-20-Screenly_OSE_4GB image.
I noticed the date on the RPI is set **Apr 20** (current date is July 2016) so I changed the start date of the assets to 2015 (just to be safe) and now I see the images on the screen.

vpetersson, 2016-07-06:
@mualemi Sounds like you don't have the current clock set then. If you have no internet connectivity, the clock cannot be set. If the clock is not set, the scheduling cannot work.