Screenly/Anthias #383: Server Error: Could not retrieve file. Check the asset URL.

thrillba, 2016-04-25:
I am getting this error with a lot of websites I try to add.  I am trying to add my google calendar, I am also trying to share an excel document through one drive.

Anyone have any ideas?  I know that the calendar issue is still an open issue and that it was kind of fixed through dropbox, but that is broken now, and no one has posted anything on it.

Is it https or what?

vpetersson, 2016-04-25:
In the case of Google Calendar, it's a different bug (#335). 

HTTPS can be a different issue, as there might be some issues with certain certificates. 

Try downloading the latest disk image (which is based on Jessie) and see if that helps.

closed 2016-04-25 by vpetersson

thrillba, 2016-04-25:
i did, just downloaded it yesterday as I have a pi 3.

vpetersson, 2016-04-28:
This has been resolved in master. Just run the upgrade script:

    $ bash <(curl -sL https://www.screenlyapp.com/install-ose.sh)

thrillba, 2016-05-16:
not sure what you have done, but its broke again.

I just did a clean install and update, and I cannot load websites again.

Please reopen