Screenly/Anthias #400: Move logging stdout/stderr

vpetersson, 2016-04-27:
With the move to systemd we should also move the logging.

Currently, we're logging to files, but since systemd already takes care of this, we should start to simply log to stdout/stderr.

vpetersson, 2016-05-02:
We also need to update [this](https://github.com/wireload/screenly-ose/blob/master/server.py#L294-L299) code.

vpetersson, 2016-05-02:
We also need to change the log format. Here's how the logging looks right now:

```
May 02 09:34:31 raspberrypi python[2229]: Mon, 02 May 2016 09:34:31 Generating asset-list...
May 02 09:34:32 raspberrypi python[2229]: Mon, 02 May 2016 09:34:32 Showing asset WireLoad (webpage)
May 02 09:34:32 raspberrypi python[2229]: Mon, 02 May 2016 09:34:32 current url is http://www.wireload.net
May 02 09:34:32 raspberrypi python[2229]: Mon, 02 May 2016 09:34:32 Sleeping for 30
May 02 09:35:03 raspberrypi python[2229]: Mon, 02 May 2016 09:35:03 Showing asset Screenly Tour (webpage)
May 02 09:35:03 raspberrypi python[2229]: Mon, 02 May 2016 09:35:03 current url is https://www.screenlyapp.com/tour/
May 02 09:35:03 raspberrypi python[2229]: Mon, 02 May 2016 09:35:03 Sleeping for 60
```

As you can see, the time stamp is redundant.

over64, 2016-05-02:
https://github.com/wireload/screenly-ose/pull/407

over64, 2016-05-02:
There is some issues:
It seems that we cannot use journalctl for this because:
1) on debian 7 it's named ```systemd-journalctl``` but on debian 8 - ```journalctl```
2) on debian 7 it does not recognize -u option for filter by unit name.
The simpliest way is use ```systemctl status screenly-web.service -n 50```. Is it ok?
It will produce something like
```
● ssh.service - OpenBSD Secure Shell server
   Loaded: loaded (/lib/systemd/system/ssh.service; enabled; vendor preset: enabled)
   Active: active (running) since Чт 2016-04-14 19:28:06 OMST; 2 weeks 3 days ago
  Process: 4645 ExecReload=/bin/kill -HUP $MAINPID (code=exited, status=0/SUCCESS)
 Main PID: 584 (sshd)
   CGroup: /system.slice/ssh.service
           └─584 /usr/sbin/sshd -D

май 02 11:22:57 over-debby14 systemd[1]: Reloading OpenBSD Secure Shell server.
май 02 11:22:57 over-debby14 sshd[584]: Received SIGHUP; restarting.
май 02 11:22:57 over-debby14 systemd[1]: Reloaded OpenBSD Secure Shell server.
май 02 11:22:57 over-debby14 sshd[584]: Server listening on 0.0.0.0 port 22.
май 02 11:22:57 over-debby14 sshd[584]: Server listening on :: port 22.
май 02 16:42:47 over-debby14 systemd[1]: Reloading OpenBSD Secure Shell server.
май 02 16:42:47 over-debby14 sshd[584]: Received SIGHUP; restarting.
май 02 16:42:47 over-debby14 systemd[1]: Reloaded OpenBSD Secure Shell server.
май 02 16:42:47 over-debby14 sshd[584]: Server listening on 0.0.0.0 port 22.
май 02 16:42:47 over-debby14 sshd[584]: Server listening on :: port 22.
```

vpetersson, 2016-05-02:
Sure, that seems reasonable.

over64, 2016-05-07:
https://github.com/wireload/screenly-ose/pull/414

vpetersson, 2016-07-12:
Confirmed working on master.

closed 2016-07-12 by vpetersson