Screenly/Anthias #412: Package libffi was not found

ross-bradbury, 2016-05-04:
Failure to install on Raspberry Pi 3 B

I had trouble upgrading a older OSE installation so I started from scratch. I erased the SD card, put on 2016-03-18-raspbian-jessie-lite, did 
sudo raspi-config (expand filesystem)
sudo reboot now

sudo apt-get update
sudo apt-get dist-upgrade
sudo apt-get install lightdm
sudo reboot now

bash <(curl -sL https://www.screenlyapp.com/install-ose.sh)

I tried installing a few times, this was from my last attempt. This attempt I answered "y" then "n"

    pi@raspberrypi:~ $ bash <(curl -sL https://www.screenlyapp.com/install-ose.sh)
    _____                           __         ____  _____ ______
    / ___/_____________  ___  ____  / /_  __   / __ \/ ___// ____/
    \__ \/ ___/ ___/ _ \/ _ \/ __ \/ / / / /  / / / /\__ \/ __/
    ___/ / /__/ /  /  __/  __/ / / / / /_/ /  / /_/ /___/ / /___
    /____/\___/_/   \___/\___/_/ /_/_/\__, /   \____//____/_____/
                                    /____/
    Screenly OSE requires a dedicated Raspberry Pi / SD card.
    You will not be able to use the regular desktop environment once installed.

    Do you still want to continue? (y/N)
    Would you like to perform a full system upgrade as well? (y/N)
    + sudo mkdir -p /etc/ansible
    + echo -e '[local]\nlocalhost ansible_connection=local'
    + sudo tee /etc/ansible/hosts
    + '[' '!' -f /etc/locale.gen ']'
    + sudo apt-get update
    Hit http://mirrordirector.raspbian.org jessie InRelease
    Hit http://archive.raspberrypi.org jessie InRelease
    Hit http://mirrordirector.raspbian.org jessie/main armhf Packages
    Hit http://archive.raspberrypi.org jessie/main armhf Packages
    Hit http://archive.raspberrypi.org jessie/ui armhf Packages
    Hit http://mirrordirector.raspbian.org jessie/contrib armhf Packages
    Hit http://mirrordirector.raspbian.org jessie/non-free armhf Packages
    Hit http://mirrordirector.raspbian.org jessie/rpi armhf Packages  
    Ign http://archive.raspberrypi.org jessie/main Translation-en_GB  
    Ign http://archive.raspberrypi.org jessie/main Translation-en     
    Ign http://archive.raspberrypi.org jessie/ui Translation-en_GB    
    Ign http://archive.raspberrypi.org jessie/ui Translation-en       
    Ign http://mirrordirector.raspbian.org jessie/contrib Translation-en_GB
    Ign http://mirrordirector.raspbian.org jessie/contrib Translation-en
    Ign http://mirrordirector.raspbian.org jessie/main Translation-en_GB
    Ign http://mirrordirector.raspbian.org jessie/main Translation-en
    Ign http://mirrordirector.raspbian.org jessie/non-free Translation-en_GB
    Ign http://mirrordirector.raspbian.org jessie/non-free Translation-en
    Ign http://mirrordirector.raspbian.org jessie/rpi Translation-en_GB                                                                                                                                                         
    Ign http://mirrordirector.raspbian.org jessie/rpi Translation-en                                                                                                                                                            
    Reading package lists... Done                                                                                                                                                                                               
    + sudo apt-get install -y python-dev git-core
    Reading package lists... Done
    Building dependency tree       
    Reading state information... Done
    git-core is already the newest version.
    python-dev is already the newest version.
    0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
    + curl -s https://bootstrap.pypa.io/get-pip.py
    + sudo python
    Requirement already up-to-date: pip in /usr/local/lib/python2.7/dist-packages
    + sudo pip install ansible==2.0.2.0
    Collecting ansible==2.0.2.0
    Collecting paramiko (from ansible==2.0.2.0)
    Using cached paramiko-2.0.0-py2.py3-none-any.whl
    Requirement already satisfied (use --upgrade to upgrade): setuptools in /usr/local/lib/python2.7/dist-packages (from ansible==2.0.2.0)
    Collecting PyYAML (from ansible==2.0.2.0)
    Collecting pycrypto>=2.6 (from ansible==2.0.2.0)
[... 232 lines cut ...]
        copying cffi/recompiler.py -> build/lib.linux-armv7l-2.7/cffi
        copying cffi/cparser.py -> build/lib.linux-armv7l-2.7/cffi
        copying cffi/gc_weakref.py -> build/lib.linux-armv7l-2.7/cffi
        copying cffi/_cffi_include.h -> build/lib.linux-armv7l-2.7/cffi
        copying cffi/parse_c_type.h -> build/lib.linux-armv7l-2.7/cffi
        copying cffi/_embedding.h -> build/lib.linux-armv7l-2.7/cffi
        running build_ext
        building '_cffi_backend' extension
        creating build/temp.linux-armv7l-2.7
        creating build/temp.linux-armv7l-2.7/c
        arm-linux-gnueabihf-gcc -pthread -DNDEBUG -g -fwrapv -O2 -Wall -Wstrict-prototypes -fno-strict-aliasing -D_FORTIFY_SOURCE=2 -g -fstack-protector-strong -Wformat -Werror=format-security -fPIC -DUSE__THREAD -I/usr/include/ffi -I/usr/include/libffi -I/usr/include/python2.7 -c c/_cffi_backend.c -o build/temp.linux-armv7l-2.7/c/_cffi_backend.o
        c/_cffi_backend.c:15:17: fatal error: ffi.h: No such file or directory
        #include <ffi.h>
                        ^
        compilation terminated.
        error: command 'arm-linux-gnueabihf-gcc' failed with exit status 1
        
        ----------------------------------------
    Command "/usr/bin/python -u -c "import setuptools, tokenize;__file__='/tmp/pip-build-WT3xOA/cffi/setup.py';exec(compile(getattr(tokenize, 'open', open)(__file__).read().replace('\r\n', '\n'), __file__, 'exec'))" install --record /tmp/pip-OUmJEr-record/install-record.txt --single-version-externally-managed --compile" failed with error code 1 in /tmp/pip-build-WT3xOA/cffi/
    + ansible localhost -m git -a 'repo=http://github.com/wireload/screenly-ose.git dest=/home/pi/screenly version=master'
    /dev/fd/63: line 54: ansible: command not found
    + cd /home/pi/screenly/ansible
    /dev/fd/63: line 55: cd: /home/pi/screenly/ansible: No such file or directory
    + ansible-playbook site.yml --skip-tags enable-ssl,system-upgrade
    /dev/fd/63: line 57: ansible-playbook: command not found

vpetersson, 2016-05-05:
I've pushed a fix for this. Please try the installer again.

Also, `sudo apt-get install lightdm` should not be needed as we've moved away from it.

closed 2016-05-05 by vpetersson

ross-bradbury, 2016-05-05:
**Thanks!** I tested this and it works great now.

With RASPBERRY PI 3 MODEL B

Erased the SD card and put on 2016-03-18-raspbian-jessie-lite
sudo raspi-config (expand filesystem)
sudo reboot now

bash <(curl -sL https://www.screenlyapp.com/install-ose.sh)
Answered both questions with "y"
Resulted in working Screenly OSE

vpetersson, 2016-05-05:
👍

commit 172288fb (2016-05-05): Closes #412. Installation hotfix.