Screenly/Anthias #413: Server Error: Could not retrieve file. Check the asset URL. when adding assets

paulbartell, 2016-05-06:
Whenever I try to add any asset by url, I get the following error:
Server Error: Could not retrieve file. Check the asset URL.

Adding assets by uploading seems to work fine though.

vpetersson, 2016-05-06:
@pastcompute Have you upgraded to the latest version? We've made some improvements there in the last week.

paulbartell, 2016-05-06:
Yes. I just upgraded this morning.

vpetersson, 2016-05-06:
Could you provide some example URLs that you have tested? I'm not able reproduce this.

paulbartell, 2016-05-06:
Hmm. You're right. Seems to only have trouble with my company's website. http://www.temperednetworks.com/wp-content/uploads/2016/05/TV-13_beer.png

vpetersson, 2016-05-09:
I was able to reproduce this. This is from the server logs:

```
May 09 10:24:15 raspberrypi python[4734]: Traceback (most recent call last):
May 09 10:24:15 raspberrypi python[4734]: File "/home/pi/screenly/server.py", line 210, in api_view
May 09 10:24:15 raspberrypi python[4734]: return make_json_response(view(*args, **kwargs))
May 09 10:24:15 raspberrypi python[4734]: File "/home/pi/screenly/server.py", line 224, in add_asset
May 09 10:24:15 raspberrypi python[4734]: raise Exception("Could not retrieve file. Check the asset URL.")
May 09 10:24:15 raspberrypi python[4734]: Exception: Could not retrieve file. Check the asset URL.
```

I was also able to reproduce this from my laptop:

```
>>> import requests
>>> r = requests.get('http://www.temperednetworks.com/wp-content/uploads/2016/05/TV-13_beer.png')
>>> r
<Response [403]>
```

We've seen this in the past. It's most likely a web application firewall (WAF). Check with your internet service provider and ask them to turn this off.

vpetersson, 2016-05-09:
I pushed an update in #413. I was unable to reproduce it after this update.

closed 2016-05-09 by vpetersson