Screenly/Anthias #423: Blank Screen

jhawk145, 2016-05-19:
When I link my companies website to the Asset, it will only give me a blank white screen. Is this some limitation of screenly based on the content of the website?
Website is https://www.teslights.com

vpetersson, 2016-05-20:
Well, that site doesn't work for me in Chrome either, so I reckon it's something wrong with the site.

closed 2016-05-20 by vpetersson

jhawk145, 2016-05-20:
Sorry about that. You are right. For some reason that one doesn't. This one should: http://teslights.com. Now when I use this link, it loads only part of the page,and loads very slowly,  and will not let me scroll.

jhawk145, 2016-05-20:
Is there a reason everything is so slow? My guess would be how much Javascript is on the page.

vpetersson, 2016-05-21:
Yeah, too much JavaScript and graphics.