Screenly/Anthias #425: another update, anther bug

rsenio, 2016-05-24:
Updated 2 screenly installs today. One seems to be working, the other displays the "screenly" splash screen at boot but does not play my assests (jpegs) just a black screen

System info just displays the below on BOTH screenly's Ive got and the screen.watchdog file is 0 kb
(no viewer log present -- is only the screenly server running?)***

rsenio, 2016-05-24:
I've also re-installed screenly and removed/re-added assets

rsenio, 2016-05-24:
The one that is working does not display jpegs, only a web page and a mp4.

Tsunayoshi, 2016-05-25:
Same here.

TASK [screenly : Run database migration] ***************************************
fatal: [localhost]: FAILED! => {"changed": true, "cmd": ["python", "/home/pi/screenly/bin/migrate.py"], "delta": "0:00:00.212002", "end": "2016-05-25 09:14:32.615726", "failed": true, "rc": 1, "start": "2016-05-25 09:14:32.403724", "stderr": "Traceback (most recent call last):\n  File \"/home/pi/screenly/bin/migrate.py\", line 158, in <module>\n    migrate_drop_filename()\n  File \"/home/pi/screenly/bin/migrate.py\", line 148, in migrate_drop_filename\n    with open_db_get_cursor() as (cursor, _):\n  File \"/usr/lib/python2.7/contextlib.py\", line 17, in __enter__\n    return self.gen.next()\n  File \"/home/pi/screenly/bin/migrate.py\", line 53, in open_db_get_cursor\n    with sqlite3.connect(database, detect_types=sqlite3.PARSE_DECLTYPES) as conn:\nsqlite3.OperationalError: unable to open database file", "stdout": "", "stdout_lines": [], "warnings": []}

rsenio, 2016-05-25:
Currently my screenly install is down since it can't play any of my jpg assets, not a good thing. Help?

Jookius, 2016-05-27:
I'm having the same problem with jpegs. Issue #278 again?

vpetersson, 2016-05-30:
@Jookius It likely is. I've reverted this. It's very strange bug.

closed 2016-05-30 by vpetersson

rsenio, 2016-05-30:
So whats the fix for this?

vpetersson, 2016-05-30:
Run the upgrade.

rsenio, 2016-05-30:
jpg's work now, but there is a black screen in between each one