Screenly/Anthias #430: Issues

rsenio, 2016-05-31:
1. screenly is showing updates are available regardless of how many times you actually update.
2. jpgs load with a black screen in between assets

Lil-briX, 2016-06-01:
Same problems here.

vpetersson, 2016-06-06:
> screenly is showing updates are available regardless of how many times you actually update.

This is a known issue. There is an update in the master branch that addresses this.

> jpgs load with a black screen in between assets

I'm unable to reproduce this myself on the master branch.

There is a new option during the installer that allows you to install the development branch (a.k.a. master) if you want to use the latest features.

closed 2016-06-06 by vpetersson

rsenio, 2016-06-06:
Installed from the master branch, black page is still present despite it not being "reproducible"

Logs show black page asset being loaded

Jun 06 09:01:57 welcomeboard python[908]: Showing asset SEA (image)
Jun 06 09:01:57 welcomeboard python[908]: current url is file:///tmp/screenly_html/black_page.html
Jun 06 09:01:57 welcomeboard python[908]: Sleeping for 8
Jun 06 09:02:05 welcomeboard python[908]: Showing asset Keysource (image)
Jun 06 09:02:05 welcomeboard python[908]: current url is file:///tmp/screenly_html/black_page.html
Jun 06 09:02:05 welcomeboard python[908]: Sleeping for 6
Jun 06 09:02:11 welcomeboard python[908]: Showing asset Combustex (image)
Jun 06 09:02:12 welcomeboard python[908]: current url is file:///tmp/screenly_html/black_page.html

vpetersson, 2016-06-06:
Please provide the output of the following:

```
$ echo 'SELECT * from assets;' | sqlite3 ~/.screenly/screenly.db
$ ls -lah ~/screenly_assets
```

rsenio, 2016-06-06:
$ echo 'SELECT * from assets;' | sqlite3 ~/.screenly/screenly.db

f0460829bf714eeb96c69afad7510a1c|Welcome|/home/pi/screenly_assets/f0460829bf714eeb96c69afad7510a1c                                                                                                                                           ||2016-05-24 19:14:00|2017-01-02 20:14:00|6|image|1||2
f9ac724cbdd14cf8932fa4dfaea7d273|Keysource|/home/pi/screenly_assets/f9ac724cbdd14cf8932fa4dfaea7d2                                                                                                                                           73||2016-05-24 19:33:00|2017-01-02 20:33:00|6|image|1||0
55cb94ae92994ff5b59c6510b21aa1be|Combustex|/home/pi/screenly_assets/55cb94ae92994ff5b59c6510b21aa1                                                                                                                                           be||2016-05-24 19:33:00|2017-01-02 20:33:00|6|image|1||1
bcc0cdec5e6d4a9ca1552e4368bb3614|SEA|/home/pi/screenly_assets/bcc0cdec5e6d4a9ca1552e4368bb3614||20                                                                                                                                           16-06-05 12:33:00|2016-06-18 16:33:00|8|image|1|0|3

ls -lah ~/screenly_assets

total 720K
drwxr-xr-x 2 pi pi 4.0K May 31 10:33 .
drwxr-xr-x 9 pi pi 4.0K Apr 27 15:00 ..
-rw-r--r-- 1 pi pi 168K May 24 13:33 55cb94ae92994ff5b59c6510b21aa1be
-rw-r--r-- 1 pi pi 190K May 31 10:33 bcc0cdec5e6d4a9ca1552e4368bb3614
-rw-r--r-- 1 pi pi 180K May 24 13:15 f0460829bf714eeb96c69afad7510a1c
-rw-r--r-- 1 pi pi 171K May 24 13:33 f9ac724cbdd14cf8932fa4dfaea7d273