Screenly/Anthias #432: Errormessage while uploading .png files

Lil-briX, 2016-06-02:
Hey, 

i have a problem while uploading .png files. 
i got 15 pictures. At first none of it worked. Then i just noticed that those pictures where in 4k resolution. So i resized them to fullhd! Now i am able to upload 14 out of those 15 pictures. 
The last one is always throwing a error message "Error. Please reload the page and try it again!"
I don't really now how to fix that. 

I would really appreciate some kind of usefull errormessage. 
Thanks alot! 

Best, 
Lil-briX

Lil-briX, 2016-06-02:
Edit: converting the same pic in 1920x1080 from .png to .gif made it work. 
I still do not understand why.

vpetersson, 2016-06-06:
I have not had any issues uploading PNGs in the past. My guess would be that it is rather related to the size (and subsequently a time out). 

Could you attach the images you're having issues with?

Also, there is really no point in uploading 4k images. You should scale them down before uploading them, as that will add a lot to the overhead of the playback (load time in particular).

Lil-briX, 2016-06-07:
Is there a logfile wich i should take a look at after uploading it? .. 
cause the errormessage don't really tell me anything

vpetersson, 2016-07-12:
This is most likely a duplicate of #444. Closing out.

closed 2016-07-12 by vpetersson