Screenly/Anthias #433: FAQ: Touch screen enabled for interactive dashboards?

bittner, 2016-06-03:
Can you add the following question / answer  to the FAQ on the website, please?

> Is Screenly touch screen enabled, so it can be used for interactive dashboards?

Knowing this would be good for use with Geckoboard and stuff, e.g. for monitoring applications.

vpetersson, 2016-06-03:
We do support dashboards, but not interactive dashboards.

In theory, it would be possible to support this, but the limiting factor is the Raspberry Pi (and partially the browser used).

If you look at something like Geckoboard, which works great with Screenly, the issues is that the regular page takes 10-30 seconds to load (depending on the amount of charts). Hence, adding interactivity would not be viable as it would be too sluggish.

closed 2016-06-03 by vpetersson

bittner, 2016-06-03:
Thanks @vpetersson for answering.

My request was to _put this into the FAQ_ section of your website.

Thank you.

vpetersson, 2016-06-03:
Well, since we don't support this, I don't think it make sense to put it on the website.

bittner, 2016-06-03:
Do you have (potential) customers asking for this feature?

If yes, please state it on the website somewhere (whether you're planning to work on it, what the limitations are, etc.), so that others don't need to do research on it. As I had to do.

Would be nice and helpful. Maybe one day this will work. Or a port of Screenly to more powerful hardware will support it (as mentioned in the FAQ, _[Can I run Screenly without a Raspberry Pi?](Can I run Screenly without a Raspberry Pi?)_).

Interactivity is definitely something that is needed for useful dashboards. Be it via voice control ("Alexa"), touch screen or both. This will be a UX factor that is simply "expected" by users in very near future. Watch sci-fi movies and convince yourself!