Screenly/Anthias #434: Error performing TLS handshake when displaying SSL secured sites.

nilsbyte, 2016-06-03:
I am using Screenly to display the website of the agency I am working for in the foyer on a big screen. Therefore I have added various pages of the website to the Screenly Asset Manager. Every once in a while Screenly fails to display the pages. From this point on, no subsequent pages are loaded. There is always an error stating:

Unable to load page

Problem occurred while loading the URL https://www.domain.com/category/

Error performing TLS handshake: GnuTLS internal error.

vpetersson, 2016-07-12:
@NiKiZe can you provide the exact URL so that we can test and see if we can reproduce it?

NiKiZe, 2016-07-12:
@vpetersson I think you meant @nilsbyte
maybe try an wget/curl on the url and see what you get, otherwise check the certificate.

vpetersson, 2016-07-13:
@NiKiZe Sorry, yes you are right. That ping was intended for @nilsbyte.

Even better than testing with wget/curl, test with this Python code (which is almost identical to how Screenly does it behind the scenes):

```
$ python
[...]
>>> import requests
>>> requests.head('http://www.google.com')
<Response [302]>
```

Replace 'google.com' with your URL.

abrenoch, 2016-09-07:
I am having this issue as well, and cannot seem to find a workaround.

I have disabled ssl verification in:
~/.screenly/screenly.conf
~/.config/uzbl/config
~/.config/uzbl/config-screenly

I have updated screenly to the latest version, and verified the time & timezones are correct.. but I still get intermittent ssl handshake failures.

I am trying to track down the best log files to reference, but I am having a difficult time locating the logs that would actually be useful to me. Could you please point me in the right direction?

Thanks!

vpetersson, 2016-09-08:
@abrenoch Try the above python snippet to see what happens.

abrenoch, 2016-09-08:
@vpetersson Thanks for the reply! I have been trying to reproduce the problem in python but have consistently been getting a valid response.

I did notice though (after updating screenly) the ~/.config/uzbl/config-screenly file did have the 'verify_ssl' enabled again. I just went in and disabled it again, so I'll report if I see an improvement with that change or not!

lenerdducdi, 2016-09-20:
Hi,
I'm having the same issue as well on my school network. It works alright at home thought.
I have the issue when :
-I try to curl the install file (https://www.screenly.io/install-ose.sh) -> curl: (35) error:140770FC:SSL routines:SSL23_GET_SERVER_HELLO: unknown protocol
-I try to display a Google Slide -> Error performing TLS handshake : An unexpected TLS packet was received

I think it might be a secured port not opened on my school's router or something related to my school network, more than a screenly issue. I tried to disable SSL as stated above without success.

Anyone has an idea on how to solve this ?

abrenoch, 2016-09-20:
@lenerdducdi Not sure this will help with your issue, but I found something that does seem to help:

`sudo update-ca-certificates`

Then a reboot seems to get uzbl working properly again, but over time the issue seems to come back (where repeating the certificate update command helps again).

lenerdducdi, 2016-09-21:
Thanks for your reply. Updating the certificates didn't help.

I tried to investigate on the ports and here's what i found :

- Screenly needs to access puppet.screenlyapp.com on port 8140 TCP, among others (as stated here : https://support.screenly.io/hc/en-us/articles/212106986-What-are-the-network-requirements-for-Screenly-).

- connecting to the other required domains and ports works, but when i try to connect to puupet.screenlyapp with `nmap -p 8140 screenlyapp.com` it gives a `STATE : filtered` instead of open.

vpetersson, 2016-09-21:
@lenerdducdi the 8140 port requirement is only for Screenly Pro. Screenly OSE does not use this port.

lenerdducdi, 2016-09-21:
@vpetersson Ok,everything seems ok with 443 ports, so do you have an idea which ones might cause a problem ?

vpetersson, 2016-09-21:
@lenerdducdi Are you using Jessie or Wheezy?

lenerdducdi, 2016-09-21:
@vpetersson Jessie, installed and updated to the latest developpment branch

lenerdducdi, 2016-09-22:
@vpetersson I tried some commands on another linux station (linux mint 18) to see if this is network or Screenly related. 

- `curl -L https://raw.githubusercontent.com/wireload/screenly-ose/master/bin/install.sh
` worked on the Linux Mint station. It showed an error on the Screenly Raspberry Pi.

- I also tried the python code you suggested above and got the following result :
```
$ python
>>> import requests
>>> requests.head('https://docs.google.com/presentation/d/1Tmx_YHUYDxhwig2XB0VbeC5Pbt4HyObX9C0endhKeOg/pub?start=false&loop=false&delayms=3000')
<Response [200]>
```
Request successful on the Linux mint station, so now I think more of an SSL certificates problem on the raspberry Pi.

vpetersson, 2017-06-30:
I'm going to close this one out, as it likely was related to an old certificate package. It has hopefully been resolved since. If not, let's re-open it.

closed 2017-06-30 by vpetersson

netjordanlee, 2018-09-25:
Requesting to reopen this as I have experienced this issue. It's easiest to replicate by setting up a script to cycle through a few different secured pages.

For reference the page I last saw the issue on is secured with the following settings:

```
Protocol: TLS 1.2
Key exchange: RSA
Cipher: AES_128_GCM
```

Unfortunately, I'm not at liberty to share the URL.

vpetersson, 2018-09-25:
@netjordanlee Can you try the experimental branch and see if that resolves it?

netjordanlee, 2018-09-28:
It is on all active branches.

greenblueberry, 2019-01-22:
I got same error in school with OSE and i think is it possible to automatically reboot raspberry when TLS problem come?

vpetersson, 2019-01-28:
I wonder what this error stems from. Is there any DPI/MiTM or proxy involved on the network?

greenblueberry, 2019-01-29:
@vpetersson not in this school internal network, link to outside network is clean

ttaureFIN, 2019-02-14:
Same issue here. Works fine for hour or two. Then jammmed

ealmonte32, 2019-02-22:
Can you guys disable the ssl verification on uzbl and viewer just to check? This seems like the same as this: #974 

edit the uzbl browser config:
`sudo nano ./.config/uzbl/config`

once in there, press ctrl+w and search for ssl, press enter
once it takes you to the line that says SSL related configuration, change the 1 in set ssl_verify to 0 so it looks like this:
`set ssl_verify = 0`
and save by pressing ctrl + x , press y, press enter..

And change in the screenly conf file if it was not already done:
`sudo nano ./.screenly/screenly.conf`

Find the line under viewer and change verify ssl to off
>[viewer]
>verify_ssl = off

and save by pressing ctrl + x , press y, press enter..

then reboot by typing `sudo reboot`

This issue is driving me crazy too so I also want to get to the bottom of it..

ealmonte32, 2019-02-23:
> vpetersson I tried some commands on another linux station (linux mint 18) to see if this is network or Screenly related.
>
> * `curl -L [...]

@lenerdducdi - when I tested this i also got error from Pi because I had the policy on the Pi to block certain websites of certain categories, but when I whitelisted the Pi in the network policy of the school firewall, any URL and the one from githubusercontent.com worked, which proved that it was a firewall issue and not a certificate issue. Again, ONLY saying this as part of the troubleshooting steps, I am not saying this is the case for your issue, there could be an issue with certificate but for sake of troubleshooting, we need to isolate by doing a top down or layer by layer approach.

morktron, 2019-04-11:
Same issue here, my modem router is using OpenDNS - https://support.opendns.com/hc/en-us/articles/228006047-Generalized-Router-Configuration-Instructions

Maybe that could be related to the issue

ealmonte32, 2019-04-11:
> Same issue here, my modem router is using OpenDNS - https://support.opendns.com/hc/en-us/articles/228006047-Generalized-Router-Configuration-Instructions
> 
> Maybe that could be related to the issue

As I've been researching this for many months now, I can safely tell you it is not your OpenDNS (I use this at work as my DNS forwarder, but not at home where I was also getting the error), and this issue seems to be the web browser UZBL (which is used to display URL Assets) and the way it handles websites with multiple certificates for the same IP.

PyOpenSSL is a "thin" wrapper around the OpenSSL library, and this package (which is used with python-requests and the version anyone uses today should certainly be 2.12+), already includes SNI support, but the browser seems to run into this TLS error sometimes when itself encounters these websites that have multiple certificates. 

**I have been able to reduce my occurrences of this error and now recently I have actually been running a few of my Screenly Pi without getting this error at all by modifying my `./.config/uzbl/config-screenly` file to be like this below:**

```
set show_status = 0
set on_event = request ON_EVENT
set cache_control = document_viewer
set fifo_dir   = /tmp
set socket_dir = /tmp

@on_event   LOAD_START    js alert=function(e){return true};confirm=function(e){return true};prompt=function(e){return true};
@on_event   LOAD_COMMIT   js alert=function(e){return true};confirm=function(e){return true};prompt=function(e){return true};
@on_event   LOAD_ERROR    exit
```

For good measure, since apps use the openssl config file for some things, I also edited my `/etc/ssl/openssl.cnf` file to make sure the certificate directory is correct.
So you can do `sudo nano /etc/ssl/openssl.cnf` and check your [ CA_default ] area for the dir location and adjust accordingly:

```
[ CA_default ]

dir             = /etc/ssl              # Where everything is kept

```
___

_This forum post shows all I've tried to get to the bottom of this:
https://forums.screenly.io/t/crash-an-unexpected-tls-packet-was-received/36/38_