Screenly/Anthias #435: Adjust period between assets?

bachma0507, 2016-06-03:
I can adjust the length of time the assets appear on the screen, but is there a way to adjust the time between the assets appearing? Right now it's about 3 seconds between assets, and I want to set it to about 1 second. Thanks

vpetersson, 2016-06-06:
@bachma0507 There is no built-in delay between assets. This is pure load time. There might be additional optimization that we could do, but it's not low hanging fruit.

closed 2016-06-06 by vpetersson

SlyLaM, 2019-04-30:
Hi !
What if we want to have a pause between assets ?
For my needs, I'd like to have an url asset that comes between all other assets (videos) but it's not possible. So I need to recreate it dozens of times to put it between every video assets in the list ...
It's very tedious work when you have many assets and when you want to change something in the list ...
The solution would be to replace the black page html with my page and to be able to introduce a delay between video assets.
Is there a way to do it ?
Thanks

vpetersson, 2019-05-01:
@SlyLaM The best way to do this is likely using the API. I'm not sure if all those functionalities are there yet, but it's not something that belongs in the UI, as it's an edge case.

Having a *video* between assets is challenging, because we can currently only *start* a video, not control it in detail. 

Code contributions are welcome however.