Screenly/Anthias #437: System sits at blank screen waiting for route when network is down.

JDogHerman, 2016-06-07:
`# wait for default route
def is_routing_up():
    try:
        sh.grep(sh.route(), 'default')
        return True
    except sh.ErrorReturnCode_1:
        return False
`

I wanted to use screenly-ose without a network always being available. The code above seems to lock screenly in a blank screen loop with errors that the playlist is not available. Only once a network connection was attached does the sh.route() able to return True. I worked around this issue by changing the `return False` to read `return True`.

Maybe a better check out be to see if the interface was up or down then wait for a route and if the interface was not up continue running the existing playlist.

vpetersson, 2016-07-12:
@over64 are you able to reproduce this?

vpetersson, 2016-07-12:
@JDogHerman Yes, it should work in offline mode. I'll let @over64 take a closer look on that.

over64, 2016-07-14:
```python
import time
import sh

# wait for default route
def is_routing_up():
    try:
        sh.grep(sh.route(), 'default')
        return True
    except sh.ErrorReturnCode_1:
        return False

for _ in range(1, 30):
    if is_routing_up():
        break
    print('Waiting for to come up...')
    time.sleep(1)
```
Hi, @JDogHerman
1) This code is used for dealyed start of screenly-web.service (systemd service). There is need because on debian 7 (jessie) + systemd we have race conditions with network services on startup. E.g  - screenly.web starts before network service(sad, but true) and it brings to "unable to resolve management IP" at splash screen. This is unacceptable and we need to deal with it.
2) On debian 8 this code is not needed, because on debian 8 network starts before user services (as expected).  But we are using it due to code base generalization
3)  This code will wait for 30 secs for network up. After 30 secs screenly web service will be started in any case.

```
[Unit]
Description=Screenly Web UI
After=network-online.target # bug on Debian 7 + systemd 44

[Service]
WorkingDirectory=/home/pi/screenly
User=pi
ExecStartPre=/usr/bin/python /home/pi/screenly/bin/wait.py
ExecStart=/usr/bin/python /home/pi/screenly/server.py
Restart=always
RestartSec=5
Environment=PYTHONPATH=/home/pi/screenly

[Install]
WantedBy=multi-user.target
```
4) And it cannot affect any things with assets playback
cheers))

seanslater, 2017-07-25:
I've been playing around with this tonight and encountered the same issue, but (as discovered in a few other issue posts), it's not actually the network stopping the playlist from running but the fact the time/date from a cold boot Pi without network access sets the time many many years in the past; so any content added while the network is connected will have today's date, and won't be shown unless the date is correct.

Easy fix was just to adjust the start date on the clips to the year 2000 - worked immediately.

Keep up the great work all :)

vpetersson, 2017-07-25:
@seanslater yeah this is because there is no Real Time Clock (RTC) on the Raspberry Pi. As such, you need to either have an internet connectivity, or use an internal NTP server to configure the clock for it to work.

closed 2017-07-25 by JDogHerman