Screenly/Anthias #441: Blank screen between assets

CueballUK, 2016-06-07:
Issue on recent update has blank screen between assets. This is more of a problem when only a single asset is displayed as the screen will blank every 10 seconds (only default timing)

CueballUK, 2016-06-07:
Here is a cut from the log. Don't ever recall the blank URL bit in the log previous to last update.
Also notice that this could be related to #217 

Jun 07 12:22:19 screenly python[791]: Showing asset Roast Dinner (image)
Jun 07 12:22:19 screenly python[791]: current url is file:///tmp/screenly_h...ml
Jun 07 12:22:19 screenly python[791]: Sleeping for 10

rsenio, 2016-06-09:
As well as https://github.com/wireload/screenly-ose/issues/430

willem2668, 2016-06-22:
+1
I have the same issue (blanks between assets) - Raspberry Pi B+ running Scrrenly OSE for more than a year, issue appeared after "ansible" update.

I prepared a new image based on "2016-04-20-Screenly_OSE_4GB.zip", this image does not show blanks between the assets, after updating the blanks are back, ...

CueballUK, 2016-07-05:
Has there been any progress on this bug? Been trying stable and unstable updates and still got blanking between assets.

rsenio, 2016-07-05:
Just black all the time when trying that viewer.py

Here's the "status" page output

● screenly-viewer.service - Screenly Viewer
Loaded: loaded (/etc/systemd/system/screenly-viewer.service; enabled)
Active: active (running) since Tue 2016-07-05 09:01:37 MDT; 41s ago
Process: 911 ExecStartPost=/bin/rm -f /tmp/screenly_html/* (code=exited, status=0/SUCCESS)
Process: 909 ExecStartPost=/bin/rm -f /tmp/uzbl_* (code=exited, status=0/SUCCESS)
Process: 905 ExecStartPre=/usr/bin/xset s noblank (code=exited, status=0/SUCCESS)
Process: 902 ExecStartPre=/usr/bin/xset -dpms (code=exited, status=0/SUCCESS)
Process: 897 ExecStartPre=/usr/bin/xset s off (code=exited, status=0/SUCCESS)
Main PID: 908 (python)
CGroup: /system.slice/screenly-viewer.service
├─908 /usr/bin/python /home/pi/screenly/viewer.py
├─951 uzbl-core --config=- --uri=file:///tmp/screenly_html/black_page.html --print-events --connect-socket /home/pi/.cache/uzbl/event_daemon
└─999 python /usr/bin/uzbl-event-manager -va start
Jul 05 09:01:50 welcomeboard python[908]: Loading browser...
Jul 05 09:01:50 welcomeboard python[908]: Browser loading file:///tmp/screenly_html/black_page.html. Running as PID 951.
Jul 05 09:01:57 welcomeboard python[908]: Generating asset-list...
Jul 05 09:01:58 welcomeboard python[908]: Showing asset Keysource (image)
Jul 05 09:01:59 welcomeboard python[908]: Sleeping for 5
Jul 05 09:02:04 welcomeboard python[908]: Showing asset Combustex (image)
Jul 05 09:02:04 welcomeboard python[908]: Sleeping for 5
Jul 05 09:02:09 welcomeboard python[908]: Showing asset Welcome (image)
Jul 05 09:02:09 welcomeboard python[908]: Sleeping for 5
Jul 05 09:02:14 welcomeboard python[908]: Showing asset Keysource (image)
Jul 05 09:02:14 welcomeboard python[908]: Sleeping for 5

vpetersson, 2016-07-06:
Maybe a silly question, but are you sure the right TTY is active?

rsenio, 2016-07-06:
I haven't changed a thing. Nor do I know how to change tty. What I ended up doing was reverting back to the image that is downloadable from  the site. It doesn't exhibit any of the black screen issue. The second its updated is when it happens.

vpetersson, 2016-07-12:
@over64 Are you able to reproduce this?

rsenio, 2016-07-12:
Should be pretty simple, apply the April image from your website to an SD card and once it's setup...update it 
bash <(curl -sL https://www.screenlyapp.com/install-ose.sh)

vpetersson, 2016-07-12:
@rsenio is your device online when you're able to reproduce this?

rsenio, 2016-07-12:
Not sure I understand the question. Online as in internet connection? Then yes, since I need to be online to update the version

vpetersson, 2016-07-13:
@rsenio Right, so the device is online while you do the upgrade of course, but I was referring to going forward once you have run the upgrade.

rsenio, 2016-07-13:
Yes, it is online all the time

CueballUK, 2016-07-20:
This occurs once the update is done via bash <(curl -sL https://www.screenlyapp.com/install-ose.sh)
It adds a blank screen html asset between each item.

'Jul 20 15:40:22 screenly-booksales python[792]: Showing asset 31 Wed Aft (image)
Jul 20 15:40:22 screenly-booksales python[792]: current url is file:///tmp/s...l
Jul 20 15:40:22 screenly-booksales python[792]: Sleeping for 10`

Where as before it wouldn't add the blank screen item.

CueballUK, 2016-07-26:
I notice this is set as non-reproducible. Can one of you try two images next to each other and when it changes do you have the next image appear or do screen go black for a second then show next image? This surely should happen to everyone and not just us.

I have three units that are all doing this. Whereas they never used to do it.

kein0r, 2016-07-27:
Same problem here. Just installed from scratch (Raspberry Pi B+)
Added first image with 10s, image is visible around 6-8 seconds followed by 2-4 seconds blank
Added two more images and a video. Images are shown again only 6-8 seconds follwed by a blank of 2-4 seconds, same for video

CueballUK, 2016-08-03:
Has this bug been forgotten about?

What could be adding this extra blank page into the playlist?

vpetersson, 2016-08-08:
@over64 Can you see if you can reproduce this?

over64, 2016-08-10:
Fixed https://github.com/wireload/screenly-ose/pull/458

vpetersson, 2016-08-10:
@CueballUK / @kein0r / @rsenio - Please upgrade to the latest development release and let us know if you can still reproduce it.

CueballUK, 2016-08-10:
Just updated and I can now confirm all three of my units are now blank screen free and transition to next image is instant.

closed 2016-08-10 by CueballUK

vpetersson, 2016-08-10:
👍