Screenly/Anthias #442: Automatic shutdown?

WTPLJim, 2016-06-13:
Hello,

After 1 week, it seems SCREENLY or the OS shuts down without reboot.
It is not responsive to remote access via browser nor can I connect via console remotely via PUTTY.

If the OS or Screenly is shutting down; can I change this to reboot?

unplugging power and plugging back in restores it to operation for a week.

Thank you

WTPLJim, 2016-06-13:
FYI: We power via transformer (NOT TV USB).
Connected via HDMI.
I've confirmed TV is online during these shutdowns.

Simply removing power / restoring power to the Pi brings it back online.

vpetersson, 2016-06-14:
> After 1 week, it seems SCREENLY or the OS shuts down without reboot.

There is no automatic shutdown built into screenly. My best guess is that it's starting to swap for one reason or another, which in turn causes the system to crash. 

If you're using heavy assets (large images or web pages) try disabling them and see if that changes the behavior.

vpetersson, 2016-07-12:
Try the following:

 * Install the latest version (i.e. run the installer and answer `y` to `Would you like to use the development branch? You will get the latest features, but things may break. (y/N)`)
 * If upgrading doesn't solve the issue, try disabling all assets and see if the issue persists. If so, it's unlikely a bug.

vpetersson, 2016-08-29:
Closing due to lack of response.

closed 2016-08-29 by vpetersson

WTPLJim, 2016-09-26:
MicroSD card failed.
Remonitoring the issue with a new 32GB microSD card.
Will advice if the new card solved the issue.