Screenly/Anthias #445: Non-locale Ansible Error on Install Script

gohanman, 2016-06-20:
This is what I'm seeing on a fresh install of raspbian jessie. I'm choosing "no" for development branch and "yes" for full system upgrade.

```
+ ansible localhost -m git -a 'repo=http://github.com/wireload/screenly-ose.git dest=/home/pi/screenly version=production'
ERROR! Unexpected Exception: (pyasn1 0.1.7 (/usr/lib/python2.7/dist-packages), Requirement.parse('pyasn1>=0.1.8'), set(['cryptography']))
the full traceback was:

Traceback (most recent call last):
  File "/usr/local/bin/ansible", line 75, in <module>
    from ansible.cli.adhoc import AdHocCLI as mycli
  File "/usr/local/lib/python2.7/dist-packages/ansible/cli/adhoc.py", line 28, in <module>
    from ansible.executor.task_queue_manager import TaskQueueManager
  File "/usr/local/lib/python2.7/dist-packages/ansible/executor/task_queue_manager.py", line 28, in <module>
    from ansible.executor.play_iterator import PlayIterator
  File "/usr/local/lib/python2.7/dist-packages/ansible/executor/play_iterator.py", line 29, in <module>
    from ansible.playbook.block import Block
  File "/usr/local/lib/python2.7/dist-packages/ansible/playbook/__init__.py", line 25, in <module>
    from ansible.playbook.play import Play
  File "/usr/local/lib/python2.7/dist-packages/ansible/playbook/play.py", line 27, in <module>
    from ansible.playbook.base import Base
  File "/usr/local/lib/python2.7/dist-packages/ansible/playbook/base.py", line 35, in <module>
    from ansible.parsing.dataloader import DataLoader
  File "/usr/local/lib/python2.7/dist-packages/ansible/parsing/dataloader.py", line 32, in <module>
    from ansible.parsing.vault import VaultLib
  File "/usr/local/lib/python2.7/dist-packages/ansible/parsing/vault/__init__.py", line 67, in <module>
    from cryptography.hazmat.primitives.hashes import SHA256 as c_SHA256
  File "/usr/local/lib/python2.7/dist-packages/cryptography/hazmat/primitives/hashes.py", line 15, in <module>
    from cryptography.hazmat.backends.interfaces import HashBackend
  File "/usr/local/lib/python2.7/dist-packages/cryptography/hazmat/backends/__init__.py", line 7, in <module>
    import pkg_resources
  File "/usr/local/lib/python2.7/dist-packages/pkg_resources/__init__.py", line 2927, in <module>
    @_call_aside
  File "/usr/local/lib/python2.7/dist-packages/pkg_resources/__init__.py", line 2913, in _call_aside
    f(*args, **kwargs)
  File "/usr/local/lib/python2.7/dist-packages/pkg_resources/__init__.py", line 2940, in _initialize_master_working_set
    working_set = WorkingSet._build_master()
  File "/usr/local/lib/python2.7/dist-packages/pkg_resources/__init__.py", line 637, in _build_master
    return cls._build_from_requirements(__requires__)
  File "/usr/local/lib/python2.7/dist-packages/pkg_resources/__init__.py", line 650, in _build_from_requirements
    dists = ws.resolve(reqs, Environment())
  File "/usr/local/lib/python2.7/dist-packages/pkg_resources/__init__.py", line 834, in resolve
    raise VersionConflict(dist, req).with_context(dependent_req)
ContextualVersionConflict: (pyasn1 0.1.7 (/usr/lib/python2.7/dist-packages), Requirement.parse('pyasn1>=0.1.8'), set(['cryptography']))
+ cd /home/pi/screenly/ansible
/dev/fd/63: line 65: cd: /home/pi/screenly/ansible: No such file or directory
+ ansible-playbook site.yml --skip-tags enable-ssl,system-upgrade
ERROR! Unexpected Exception: (pyasn1 0.1.7 (/usr/lib/python2.7/dist-packages), Requirement.parse('pyasn1>=0.1.8'), set(['cryptography']))
the full traceback was:

Traceback (most recent call last):
  File "/usr/local/bin/ansible-playbook", line 73, in <module>
    mycli = getattr(__import__("ansible.cli.%s" % sub, fromlist=[myclass]), myclass)
  File "/usr/local/lib/python2.7/dist-packages/ansible/cli/playbook.py", line 30, in <module>
    from ansible.executor.playbook_executor import PlaybookExecutor
  File "/usr/local/lib/python2.7/dist-packages/ansible/executor/playbook_executor.py", line 27, in <module>
    from ansible.executor.task_queue_manager import TaskQueueManager
  File "/usr/local/lib/python2.7/dist-packages/ansible/executor/task_queue_manager.py", line 28, in <module>
    from ansible.executor.play_iterator import PlayIterator
  File "/usr/local/lib/python2.7/dist-packages/ansible/executor/play_iterator.py", line 29, in <module>
    from ansible.playbook.block import Block
  File "/usr/local/lib/python2.7/dist-packages/ansible/playbook/__init__.py", line 25, in <module>
    from ansible.playbook.play import Play
  File "/usr/local/lib/python2.7/dist-packages/ansible/playbook/play.py", line 27, in <module>
    from ansible.playbook.base import Base
  File "/usr/local/lib/python2.7/dist-packages/ansible/playbook/base.py", line 35, in <module>
    from ansible.parsing.dataloader import DataLoader
  File "/usr/local/lib/python2.7/dist-packages/ansible/parsing/dataloader.py", line 32, in <module>
    from ansible.parsing.vault import VaultLib
  File "/usr/local/lib/python2.7/dist-packages/ansible/parsing/vault/__init__.py", line 67, in <module>
    from cryptography.hazmat.primitives.hashes import SHA256 as c_SHA256
  File "/usr/local/lib/python2.7/dist-packages/cryptography/hazmat/primitives/hashes.py", line 15, in <module>
    from cryptography.hazmat.backends.interfaces import HashBackend
  File "/usr/local/lib/python2.7/dist-packages/cryptography/hazmat/backends/__init__.py", line 7, in <module>
    import pkg_resources
  File "/usr/local/lib/python2.7/dist-packages/pkg_resources/__init__.py", line 2927, in <module>
    @_call_aside
  File "/usr/local/lib/python2.7/dist-packages/pkg_resources/__init__.py", line 2913, in _call_aside
    f(*args, **kwargs)
  File "/usr/local/lib/python2.7/dist-packages/pkg_resources/__init__.py", line 2940, in _initialize_master_working_set
    working_set = WorkingSet._build_master()
  File "/usr/local/lib/python2.7/dist-packages/pkg_resources/__init__.py", line 637, in _build_master
    return cls._build_from_requirements(__requires__)
  File "/usr/local/lib/python2.7/dist-packages/pkg_resources/__init__.py", line 650, in _build_from_requirements
    dists = ws.resolve(reqs, Environment())
  File "/usr/local/lib/python2.7/dist-packages/pkg_resources/__init__.py", line 834, in resolve
    raise VersionConflict(dist, req).with_context(dependent_req)
ContextualVersionConflict: (pyasn1 0.1.7 (/usr/lib/python2.7/dist-packages), Requirement.parse('pyasn1>=0.1.8'), set(['cryptography']))
```

Manually running `sudo pip install pyasn1==0.1.8` and then re-running ose-install.sh does work. At a glance this appears to be either an ansible or pip problem in how dependencies are declared but I'm far from an expert in either system.

vpetersson, 2016-06-20:
So there's a bug in the Raspbian image where it doesn't properly set the locale. You can fix this by running `raspi-config` and set the locale. I would recommend that you set it to `en_US.UTF-8 UTF-8`. 

We [try](https://github.com/wireload/screenly-ose/blob/master/bin/install.sh#L50-L54) to automatically resolve this, but it doesn't always work. 

Try the above and hopefully that will resolve the issue and you can successfully run the installer.

gohanman, 2016-06-20:
It's **not** the locale issue. I'd already done that and I've now seen the same issue on two different pis both of which had the locale set to `en_US.UTF-8 UTF-8`. The `pip install ansible==2.0.2.0` step installs `pyasn1` version `0.1.7` and then blows up with the above stack trace because it actually needs a version `>=0.1.8`.

The install script will continually fail with this error even with locale set correctly until I run `pip install pyasn1==0.1.8` to manually work around whatever's going wrong with dependency resolution.

I'm noticing github doesn't make it easy to read, but with extra newlines this seems to be the root issue:
```
Unexpected Exception: 
  (pyasn1 0.1.7 (/usr/lib/python2.7/dist-packages), 
  Requirement.parse('pyasn1>=0.1.8'), 
  set(['cryptography'])
  )
```

I think this is really `pip`'s fault rather than screenly's but nevertheless creates problems for the install script.

vpetersson, 2016-06-21:
Interesting find. I've bumped up Ansible to 2.1. Not sure if that resolves the issue for you. We do not however hardcore the version of `pyasn1`, so that must be upstream. I guess we could hardcode it to 0.1.8. but hopefully simply bumping up the Ansible version will do the trick.

gohanman, 2016-06-21:
Follow-up: I have two more pis to provision with screenly. The first one shows the same error w/ Ansible 2.1 and manually installing `pyasn1==0.1.8` still fixes it. Should the last one behave differently I'll post an update but I'm not holding my breath.

I definitely agree that it looks like an upstream problem. I'm just not familiar enough with Ansible & pip to know where it should be reported.

vpetersson, 2016-06-21:
@gohanman I would recommend that you file an ticket with Ansible [here](https://github.com/ansible/ansible/issues) and then link back to this issue.

Aketzu, 2016-07-03:
"Me too"... Just did update on old screenly and ran into this issue. Installing pyasn1==0.1.8 fixed it.

Debian has python-pyasn1 package installed which is version 0.1.7. It might be that either pip doesn't want to update system packages or it gets confused by paramiko and cryptography both having dependency on pyasn1.

Quick workaround would be to remove system package before ansible installation, i.e. add `apt-get -y remove python-pyasn1` to install-ose.sh

vpetersson, 2016-07-04:
@Aketzu Good point. 

I've pushed a fix to the development branch (master) that should hopefully solve the issue.

closed 2016-07-04 by vpetersson