Screenly/Anthias #449: Screenly OSE PHP and What Languages Supports

yasar9914, 2016-06-29:
Hi all, 

I was just trying to connect with mysql from screenly ose, but the php is not getting installed. Could you please suggest what is the best programming language to connect with database. Thanks in advance.

vpetersson, 2016-06-29:
Screenly OSE is written in Python. The database used behind the scenes is SQLite. 

I'm not sure what you're looking to do, but if you want to extend the core of Screenly, you would have to do that in Python as we would not want to mix languages.

closed 2016-06-29 by vpetersson

yasar9914, 2016-06-29:
Thanks for your answer.

It was an idea that, my php page is not displaying in the browser. If it
successfully displayed, then probably I can connect with mysql database,
which allows me to change data n a real time within raspberry pi.

But php is not getting installed with screenly.

Will php get installed in screenly ?

vpetersson, 2016-06-29:
While you could install PHP *on* the Raspberry Pi, I would strongly advice against it. Installing MySQL on the device would probably cause significant performance degradation. 

A better solution would be to simply host the PHP website somewhere else and simply add this as an asset to Screenly.

yasar9914, 2016-06-29:
Yes Victor, I have understand. It would be really a great help.

My Last Question : So May I ask whether internet is a must to execute ?

vpetersson, 2016-06-29:
Normally, no. If the web server you're pointing to where the PHP app is running, then yes. If you have it on the same LAN, then no.