Screenly/Anthias #451: Change asset-list generation rate?

blackrabt, 2016-07-03:
Could we get an option in settings to have screenlyOSE check for new assets to add to the playlist more frequently than the default? I saw that it was every Nth cycles in the pi forum but my application would benefit from more control over when things get loaded from the list of active items. My current testing shows it can be over 5 hours if you do not reboot to prompt the generation, even with a short playlist.

blackrabt, 2016-07-03:
If that doesn't suit the project I would be happy if I could just get pointed to where in the code I could go to change the rate.

vpetersson, 2016-07-04:
@blackrabt I'm not sure where you how you came to that conclusion. The playlist is re-read upon modification ([code](https://github.com/wireload/screenly-ose/blob/master/viewer.py#L88-L90)). Hence, as soon as the local database is modified, the playlist is refreshed).  This is checked between every asset.

closed 2016-07-04 by vpetersson

blackrabt, 2016-07-04:
@vpetersson  Well I have had it running all day and it has not been adding in assets when they hit active times. I guess I am explaining myself badly.

vpetersson, 2016-07-04:
Check the time of the Raspberry Pi. If it doesn't have the correct time, that could explain the issue.

blackrabt, 2016-07-04:
Will do. Thanks for assisting me.

blackrabt, 2016-07-04:
@vpetersson I wiped the card, reinstalled from the ScreenlyOSE image, ran a system update and the screenly updater. System time was accurate to within 2 minutes and for some reason I am still having problems with assets not loading in properly. I set it so that new ones would be active and then one taken off a few minutes later to try and prompt a database change but no luck. Any other suggestions?

vpetersson, 2016-07-04:
Did you use the stable or development branch?

blackrabt, 2016-07-04:
This was on the development branch.

vpetersson, 2016-07-04:
Just to make sure it isn't running on the old branch -- try restarting the device.

blackrabt, 2016-07-04:
Tried that twice sir. To give you a little more detail, it is only loading assets with a start time that is before the current time.

blackrabt, 2016-07-05:
@vpetersson I saved that test piece to an image to come back to later. I did a fresh install of the non development branch and am still having this issue. I tried another test with it online and the problem seems to go away. So can you suggest what I need to do to make this work offline please?

vpetersson, 2016-07-05:
If it is an offline issue, it's a duplicate of #247 which should have been resolved some time ago. Will continue the discussion there.