Screenly/Anthias #453: black screen but active asset

magnets99, 2016-07-06:
there are a few issues that are similar, but none with the same symptoms:
I load tomorrows image (local) and set it to come on at 5:46am however when i get into work, the screen is just showing a black screen.
When i log into screenlys web interface the schedule overview show the asset is active, but the system info page shows the playlist is empty.

[image]

[image]

The clock is correct. The pi is powered 24/7, i've tried swapping the power supply and rpi (i am still using the same sd card but it has been rebuilt).

I am using a cron job to reboot the pi, but it does this before the screen is active (by an hour)

i'm not sure how to get the log you need, if you let me know the file location i'll get it.

vpetersson, 2016-07-12:
I can see that you're running an older version. Please try install the latest version (i.e. run the installer and answer `y` to `Would you like to use the development branch? You will get the latest features, but things may break. (y/N)`)

magnets99, 2016-07-12:
I'll give it a go, thanks! (i'll come back and update). I've also changed the sd card and it rebuilt it and it hasn't done it since then. But i'm interested in the new version.

magnets99, 2016-07-18:
it hasn't done it since (i haven't updated as that's a bit of a pain given my network security) but i think it's likely to have been my network being awkward.

closed 2016-07-18 by magnets99

vpetersson, 2016-07-18:
Understood. Thanks for the update.