Screenly/Anthias #463: last shown image hangs if asset list was cleared

over64, 2016-08-12:
[image]

IRTermite, 2016-09-21:
I've also seen it force a reboot on this hang... done it a number of times. Guess that is a watchdog function?

vpetersson, 2016-09-21:
@IRTermite We are using the watchdog device on the Raspberry Pi. The watchdog however is triggered on CPU load (if the load average exceeds certain levels). This may or may not be related to this bug.

over64, 2017-01-09:
@IRTermite, @vpetersson it's scheduler-related

vpetersson, 2017-01-09:
@over64 cool. Alright. It's in this sprint so let's sort it out.

antonmolodykh, 2017-03-29:
@vpetersson the problem is related to the fact that the loading.jpg is missed

vpetersson, 2017-03-29:
@antonmolodykh Interesting. Where is `loading.jpg` used?

Regardless, let's create a pull request and assign it to this Issue.

antonmolodykh, 2017-03-29:
@vpetersson loading.jpg is used in the asset_loop function in viewer.py, in case when the list of asset is empty. I mean that loading desired picture into the project directory is enough to solve the problem (I do not know which picture)

vpetersson, 2017-06-29:
I'm not sure what happened to `loading.jpg`. It used to be the image we displayed when the playlist was empty. I've created a new one (`loading.png`). Let's implement it instead. 

[image]

closed 2017-06-29 by vpetersson