Screenly/Anthias #477: 'module' object has no attribute 'codegen'

vpetersson, 2016-08-27:
I tried the latest master branch today and it's not starting. 

Here's what the logs are saying:

[image]

over64, 2016-08-29:
It's strange, but master branch contains requirements.txt with pyhaml of 1.1.0 version. But in local repo I have pyhaml with 0.1.9 version. I changed version to 1.1.0 when first PR for email-signup was made. After this I reverted version back to 0.1.9 because update was not needed. BUT something happened bettwen pull requests reopen and merges. Need to revert pyhaml version to 0.1.9 in master

vpetersson, 2016-08-31:
Fixed in #484.

closed 2016-08-31 by vpetersson