Screenly/Anthias #478: Older installations of Screenly vulnerable to MITM

leonelwilliams, 2016-08-29:
Hi, when you moved to https://www.screenly.io a 301 redirect was put in place redirecting  from `https://www.screenlyapp.com/*` to `http://www.screenlyapp.com/`.

The `run_upgrade.sh` script in somewhat older versions of Screenly OSE will then follow this redirect chain: `https://www.screenlyapp.com/install-ose.sh` -> `http://www.screenly.io/install-ose.sh` -> `https://www.screenly.io/install-ose.sh`. The issue here is the `http://` in the middle -- if there were an MITM they could intercept the redirection to HTTPS and push their own malicious code down the pipe.

I'd recommending changing the redirect to use `https://` and using the curl option `--proto '=https'` in `run_upgrade.sh` to force curl to never use protocols other than HTTPS so this can't happen again.

vpetersson, 2016-08-29:
Hi @leonelwilliams. That's a good point. We'll look into it.

leonelwilliams, 2016-08-30:
Thanks! Though it turns out I was mistaken -- the current version still uses https://www.screenlyapp.com/install-ose.sh so this would affect all versions

vpetersson, 2016-08-30:
Fixed.

```
➜  ~ curl -IL https://www.screenlyapp.com/install-ose.sh
HTTP/1.1 301 Moved Permanently
Date: Tue, 30 Aug 2016 14:35:39 GMT
Connection: keep-alive
Set-Cookie: __cfduid=df75ce460cbd8a27ba90469c243788e0b1472567739; expires=Wed, 30-Aug-17 14:35:39 GMT; path=/; domain=.screenlyapp.com; HttpOnly
Location: https://www.screenly.io/install-ose.sh
Server: cloudflare-nginx
CF-RAY: 2da8f872e5d113a1-LHR

HTTP/1.1 302 Moved Temporarily
Date: Tue, 30 Aug 2016 14:35:39 GMT
Content-Type: text/html
Connection: keep-alive
Set-Cookie: __cfduid=d8c62d8e609eba17a28590f366ddc6c471472567739; expires=Wed, 30-Aug-17 14:35:39 GMT; path=/; domain=.screenly.io; HttpOnly
Location: https://raw.github.com/wireload/screenly-ose/master/bin/install.sh
Server: cloudflare-nginx
CF-RAY: 2da8f8742b213470-LHR

HTTP/1.1 301 Moved Permanently
Location: https://raw.githubusercontent.com/wireload/screenly-ose/master/bin/install.sh
Content-Length: 0
Accept-Ranges: bytes
Date: Tue, 30 Aug 2016 14:35:40 GMT
Via: 1.1 varnish
Age: 0
Connection: keep-alive
X-Served-By: cache-lhr6338-LHR
X-Cache: MISS
X-Cache-Hits: 0
Vary: Accept-Encoding
X-Fastly-Request-ID: f6988e293184d4b06fe83b372b9c0b6671763b07

HTTP/1.1 200 OK
Content-Security-Policy: default-src 'none'; style-src 'unsafe-inline'
Strict-Transport-Security: max-age=31536000
X-Content-Type-Options: nosniff
X-Frame-Options: deny
X-XSS-Protection: 1; mode=block
ETag: "3ecf57e2583b3c3938f3407241f628925e275ca4"
Content-Type: text/plain; charset=utf-8
Cache-Control: max-age=300
X-Geo-Block-List:
X-GitHub-Request-Id: B91F1322:3D52:9A76A57:57C598FD
Content-Length: 2893
Accept-Ranges: bytes
Date: Tue, 30 Aug 2016 14:35:40 GMT
Via: 1.1 varnish
Connection: keep-alive
X-Served-By: cache-lhr6336-LHR
X-Cache: HIT
X-Cache-Hits: 1
Vary: Authorization,Accept-Encoding
Access-Control-Allow-Origin: *
X-Fastly-Request-ID: 1ff5263ab6ae1c994d00fb7366449eefe0a71888
Expires: Tue, 30 Aug 2016 14:40:40 GMT
Source-Age: 191
```

closed 2016-08-30 by vpetersson

leonelwilliams, 2016-08-30:
Thanks! I've created #482 and #483 PRs