Screenly/Anthias #479: Black Screen after update

firepro20, 2016-08-30:
I have updated the screenly ose following the instructions provided on the website. 
I used the `bash <(curl -sL https://www.screenly.io/install-ose.sh)`

This is the failing procedure:

`TASK [system : Install Screenly dependencies] **********************************
failed: [localhost] (item=[u'console-data', u'libffi-dev', u'libssl-dev', u'matchbox', u'omxplayer', u'python-dev', u'python-netifaces', u'python-simplejson', u'rpi-update', u'sqlite3', u'systemd', u'uzbl', u'x11-xserver-utils', u'xserver-xorg']) => {"cache_update_time": 0, "cache_updated": false, "failed": true, "item": ["console-data", "libffi-dev", "libssl-dev", "matchbox", "omxplayer", "python-dev", "python-netifaces", "python-simplejson", "rpi-update", "sqlite3", "systemd", "uzbl", "x11-xserver-utils", "xserver-xorg"], "msg": "'/usr/bin/apt-get -y -o \"Dpkg::Options::=--force-confdef\" -o \"Dpkg::Options::=--force-confold\"     install 'rpi-update'' failed: E: Sub-process /usr/bin/dpkg returned an error code (1)\n", "stderr": "E: Sub-process /usr/bin/dpkg returned an error code (1)\n", "stdout": "Reading package lists...\nBuilding dependency tree...\nReading state information...\nThe following NEW packages will be installed:\n  rpi-update\n0 upgraded, 1 newly installed, 0 to remove and 0 not upgraded.\n1 not fully installed or removed.\nNeed to get 4408 B of archives.\nAfter this operation, 45.1 kB of additional disk space will be used.\nGet:1 http://archive.raspberrypi.org/debian/ jessie/main rpi-update all 20140705 [4408 B]\nFetched 4408 B in 0s (26.9 kB/s)\nSetting up install-info (5.2.0.dfsg.1-6) ...\r\nNot a directory: /usr/share/info.\r\ndpkg: error processing package install-info (--configure):\r\n subprocess installed post-installation script returned error exit status 1\r\nErrors were encountered while processing:\r\n install-info\r\n", "stdout_lines": ["Reading package lists...", "Building dependency tree...", "Reading state information...", "The following NEW packages will be installed:", "  rpi-update", "0 upgraded, 1 newly installed, 0 to remove and 0 not upgraded.", "1 not fully installed or removed.", "Need to get 4408 B of archives.", "After this operation, 45.1 kB of additional disk space will be used.", "Get:1 http://archive.raspberrypi.org/debian/ jessie/main rpi-update all 20140705 [4408 B]", "Fetched 4408 B in 0s (26.9 kB/s)", "Setting up install-info (5.2.0.dfsg.1-6) ...", "Not a directory: /usr/share/info.", "dpkg: error processing package install-info (--configure):", " subprocess installed post-installation script returned error exit status 1", "Errors were encountered while processing:", " install-info"]}
`

I reboot the Raspberry Pi 2 Model B and cannot access through IP aaa.bbb.ccc.ddd:8080

vpetersson, 2016-08-30:
Try:
```
sudo dpkg-reconfigure -a
```

firepro20, 2016-08-30:
`sudo dpkg-reconfigure -a
Unknown option: a
Usage: dpkg-reconfigure [options] packages
  -u,  --unseen-only            Show only not yet seen questions.
       --default-priority       Use default priority instead of low.
       --force                  Force reconfiguration of broken packages.
       --no-reload              Do not reload templates. (Use with caution.)
  -f,  --frontend               Specify debconf frontend to use.
  -p,  --priority               Specify minimum priority question to show.
       --terse                  Enable terse mode.
`

vpetersson, 2016-08-30:
My bad. Should have been:

```
sudo dpkg --configure -a
```

firepro20, 2016-08-30:
`
Setting up install-info (5.2.0.dfsg.1-6) ...
Not a directory: /usr/share/info.
dpkg: error processing package install-info (--configure):
 subprocess installed post-installation script returned error exit status 1
Errors were encountered while processing:
 install-info

`

vpetersson, 2016-08-30:
Seems like something is broken in the system. Perhaps the SD card is toast.

I'd suggest re-imaging.

closed 2016-08-30 by vpetersson

firepro20, 2016-08-30:
Yes that's what I'm doing will close if I fix it