Screenly/Anthias #492: See 442 - video blacks out after time

WTPLJim, 2016-10-05:
After #442, replaced mSDHC card (failed).
After 9 days or so (9/26 - 10/5 uptime) the screen blanked overnight.

I was able to access the GUI on 8080.
I SSH'd to OS and executed SUDO reboot.

Screenly worked fine upon boot.

Suggestions?

vpetersson, 2016-10-05:
Hard to tell without logs. Please post the logs output of `sudo journalctl` next time it happens.

WTPLJim, 2016-10-05:
[piLogs.txt](https://github.com/wireload/screenly-ose/files/511417/piLogs.txt)
...I see some failures at 13:11:23 ending with setting a screen mode.

Oct 05 13:11:23 raspberrypi systemd[224]: Failed at step EXEC spawning /etc/init.d/screenly_net_mgr: Exec format error
Oct 05 13:11:23 raspberrypi systemd[1]: screenly_net_mgr.service: control process exited, code=exited status=203
Oct 05 13:11:23 raspberrypi systemd[1]: Failed to start LSB: Screenly network config tool.
Oct 05 13:11:23 raspberrypi systemd[1]: Unit screenly_net_mgr.service entered failed state.
Oct 05 13:11:23 raspberrypi raspi-config[221]: Checking if shift key is held down:Error opening '/dev/input/event*': No such file or directory
Oct 05 13:11:23 raspberrypi raspi-config[221]: No. Switching to ondemand scaling governor.
Oct 05 13:11:23 raspberrypi kbd[223]: Setting console screen modes.
Oct 05 13:11:23 raspberrypi systemd-journal[133]: Runtime journal is using 5.7M (max allowed 46.2M, trying to leave 69.4M free of 456.4M available → current
Oct 05 13:11:23 raspberrypi kbd[223]: setterm: $TERM is not defined.

I can post full logs in, perhaps, a week (when I expect it to recur). If the above helps, great!

vpetersson, 2016-10-05:
That should have been fixed in the latest branch. Try running the upgrade and install the development version.

WTPLJim, 2016-10-05:
Ran update successfully. 
Will continue to monitor and watch logs.

WTPLJim, 2016-10-14:
Repeated, noticed last night. journalctl is long, mostly display of objects... Below are noted issues from the log. After this screen blanks, logs show display of items. Performing a sudo reboot to restore.

Oct 13 00:01:21 raspberrypi CRON[441]: pam_unix(cron:session): session opened for user root by (uid=0)
Oct 13 00:01:38 raspberrypi CRON[449]: (root) CMD (/usr/bin/python /usr/sbin/screenly_net_watchdog.py)
Oct 13 00:02:18 raspberrypi kernel: INFO: task kworker/u8:1:32517 blocked for more than 120 seconds.
Oct 13 00:02:28 raspberrypi kernel:       Not tainted 4.4.21-v7+ #911
Oct 13 00:02:31 raspberrypi kernel: "echo 0 > /proc/sys/kernel/hung_task_timeout_secs" disables this message.
Oct 13 00:02:31 raspberrypi kernel: kworker/u8:1    D 805b5e84     0 32517      2 0x00000000
Oct 13 00:02:31 raspberrypi kernel: Workqueue: kmmcd mmc_rescan
Oct 13 00:02:31 raspberrypi kernel: [<805b5e84>] (__schedule) from [<805b63fc>] (schedule+0x50/0xa8)
Oct 13 00:02:31 raspberrypi kernel: [<805b63fc>] (schedule) from [<8046c470>] (__mmc_claim_host+0xb8/0x1cc)
Oct 13 00:02:31 raspberrypi kernel: [<8046c470>] (__mmc_claim_host) from [<8046c5b4>] (mmc_get_card+0x30/0x34)
Oct 13 00:02:31 raspberrypi kernel: [<8046c5b4>] (mmc_get_card) from [<80474688>] (mmc_sd_detect+0x2c/0x80)
Oct 13 00:02:31 raspberrypi kernel: [<80474688>] (mmc_sd_detect) from [<8046ebc4>] (mmc_rescan+0xc8/0x324)
Oct 13 00:02:31 raspberrypi kernel: [<8046ebc4>] (mmc_rescan) from [<8003c720>] (process_one_work+0x154/0x458)
Oct 13 00:02:31 raspberrypi kernel: [<8003c720>] (process_one_work) from [<8003ca78>] (worker_thread+0x54/0x500)
Oct 13 00:02:31 raspberrypi kernel: [<8003ca78>] (worker_thread) from [<80042754>] (kthread+0xec/0x104)
Oct 13 00:02:31 raspberrypi kernel: [<80042754>] (kthread) from [<8000fbc8>] (ret_from_fork+0x14/0x2c)
Oct 13 00:03:14 raspberrypi python[2803]: Showing asset ToSmugMugScreenly (webpage)
Oct 13 00:03:14 raspberrypi python[2803]: Sleeping for 10
Oct 13 00:06:19 raspberrypi kernel: INFO: task kworker/u8:1:32517 blocked for more than 120 seconds.
Oct 13 00:06:21 raspberrypi kernel:       Not tainted 4.4.21-v7+ #911
Oct 13 00:06:21 raspberrypi kernel: "echo 0 > /proc/sys/kernel/hung_task_timeout_secs" disables this message.
Oct 13 00:06:21 raspberrypi kernel: kworker/u8:1    D 805b5e84     0 32517      2 0x00000000
Oct 13 00:06:21 raspberrypi kernel: Workqueue: kmmcd mmc_rescan
Oct 13 00:06:21 raspberrypi kernel: [<805b5e84>] (__schedule) from [<805b63fc>] (schedule+0x50/0xa8)
Oct 13 00:06:21 raspberrypi kernel: [<805b63fc>] (schedule) from [<8046c470>] (__mmc_claim_host+0xb8/0x1cc)
Oct 13 00:06:21 raspberrypi kernel: [<8046c470>] (__mmc_claim_host) from [<8046c5b4>] (mmc_get_card+0x30/0x34)
Oct 13 00:06:21 raspberrypi kernel: [<8046c5b4>] (mmc_get_card) from [<80474688>] (mmc_sd_detect+0x2c/0x80)
Oct 13 00:06:21 raspberrypi kernel: [<80474688>] (mmc_sd_detect) from [<8046ebc4>] (mmc_rescan+0xc8/0x324)
Oct 13 00:06:21 raspberrypi kernel: [<8046ebc4>] (mmc_rescan) from [<8003c720>] (process_one_work+0x154/0x458)
Oct 13 00:06:21 raspberrypi kernel: [<8003c720>] (process_one_work) from [<8003ca78>] (worker_thread+0x54/0x500)
Oct 13 00:06:21 raspberrypi kernel: [<8003ca78>] (worker_thread) from [<80042754>] (kthread+0xec/0x104)
Oct 13 00:06:21 raspberrypi kernel: [<80042754>] (kthread) from [<8000fbc8>] (ret_from_fork+0x14/0x2c)
Oct 13 00:08:07 raspberrypi python[2803]: Asset ToSmugMugScreenly at https://whitehallpl.smugmug.com/frame/slideshow?key=cT6vsv&autoStart=1&captions=0 is not available, skipping.
Oct 13 00:08:18 raspberrypi kernel: INFO: task kworker/u8:1:32517 blocked for more than 120 seconds.
Oct 13 00:08:26 raspberrypi kernel:       Not tainted 4.4.21-v7+ #911
Oct 13 00:08:35 raspberrypi kernel: "echo 0 > /proc/sys/kernel/hung_task_timeout_secs" disables this message.
Oct 13 00:08:43 raspberrypi kernel: kworker/u8:1    D 805b5e84     0 32517      2 0x00000000
Oct 13 00:08:52 raspberrypi kernel: Workqueue: kmmcd mmc_rescan
Oct 13 00:09:00 raspberrypi kernel: [<805b5e84>] (__schedule) from [<805b63fc>] (schedule+0x50/0xa8)
Oct 13 00:09:10 raspberrypi kernel: [<805b63fc>] (schedule) from [<8046c470>] (__mmc_claim_host+0xb8/0x1cc)
Oct 13 00:09:15 raspberrypi kernel: [<8046c470>] (__mmc_claim_host) from [<8046c5b4>] (mmc_get_card+0x30/0x34)
Oct 13 00:09:17 raspberrypi kernel: [<8046c5b4>] (mmc_get_card) from [<80474688>] (mmc_sd_detect+0x2c/0x80)
Oct 13 00:09:21 raspberrypi kernel: [<80474688>] (mmc_sd_detect) from [<8046ebc4>] (mmc_rescan+0xc8/0x324)
Oct 13 00:09:21 raspberrypi kernel: [<8046ebc4>] (mmc_rescan) from [<8003c720>] (process_one_work+0x154/0x458)
Oct 13 00:09:21 raspberrypi kernel: [<8003c720>] (process_one_work) from [<8003ca78>] (worker_thread+0x54/0x500)
Oct 13 00:09:21 raspberrypi kernel: [<8003ca78>] (worker_thread) from [<80042754>] (kthread+0xec/0x104)
Oct 13 00:09:21 raspberrypi kernel: [<80042754>] (kthread) from [<8000fbc8>] (ret_from_fork+0x14/0x2c)
Oct 13 00:09:25 raspberrypi python[2803]: Showing asset ToSmugMugScreenly (webpage)
Oct 13 00:09:25 raspberrypi python[2803]: Sleeping for 10
Oct 13 00:09:37 raspberrypi screenly_net_watchdog.py[451]: Starting net_watchdog.
Oct 13 00:09:39 raspberrypi python[2803]: Showing asset ToSmugMugScreenly (webpage)
Oct 13 00:09:39 raspberrypi python[2803]: Sleeping for 10
Oct 13 00:09:39 raspberrypi CRON[441]: pam_unix(cron:session): session closed for user root
Oct 13 00:09:50 raspberrypi python[2803]: Showing asset ToSmugMugScreenly (webpage)
Oct 13 00:09:50 raspberrypi python[2803]: Sleeping for 10
Oct 13 00:10:00 raspberrypi python[2803]: Showing asset ToSmugMugScreenly (webpage)
Oct 13 00:10:00 raspberrypi python[2803]: Sleeping for 10
Oct 13 00:10:11 raspberrypi python[2803]: Showing asset ToSmugMugScreenly (webpage)
Oct 13 00:10:11 raspberrypi python[2803]: Sleeping for 10
Oct 13 00:10:21 raspberrypi python[2803]: Showing asset ToSmugMugScreenly (webpage)

WTPLJim, 2016-10-14:
The object is the only object in the queue to display. 
It looks like when it blocks (perhaps due to network availability?) it does not attempt to restore.

WTPLJim, 2016-10-14:
Possible occurance to prior issue 100?

WTPLJim, 2016-10-14:
I suspect, perhaps the network goes down somewhere upstream of my gateway.
I have 1 asset, a webpage where images are fed into.
Screenly sees this as unavailable - and blanks the display.

https://github.com/wireload/screenly-ose/commit/9dcc407315fa9ff738d994750902adb35f66aac1

         if asset is None:
             # The playlist is empty, go to sleep.

vpetersson, 2016-10-21:
@over64 can you reproduce this?

over64, 2016-10-22:
due to logs there is hardware problem with IO
```
Oct 13 00:08:18 raspberrypi kernel: INFO: task kworker/u8:1:32517 blocked for more than 120 seconds.
Oct 13 00:08:26 raspberrypi kernel: Not tainted 4.4.21-v7+ #911
Oct 13 00:08:35 raspberrypi kernel: "echo 0 > /proc/sys/kernel/hung_task_timeout_secs" disables this message.
Oct 13 00:08:43 raspberrypi kernel: kworker/u8:1 D 805b5e84 0 32517 2 0x00000000
Oct 13 00:08:52 raspberrypi kernel: Workqueue: kmmcd mmc_rescan
Oct 13 00:09:00 raspberrypi kernel: <805b5e84> from <805b63fc>
Oct 13 00:09:10 raspberrypi kernel: <805b63fc> from <8046c470>
Oct 13 00:09:15 raspberrypi kernel: <8046c470> from <8046c5b4>
Oct 13 00:09:17 raspberrypi kernel: <8046c5b4> from <80474688>
Oct 13 00:09:21 raspberrypi kernel: <80474688> from <8046ebc4>
Oct 13 00:09:21 raspberrypi kernel: <8046ebc4> from <8003c720>
Oct 13 00:09:21 raspberrypi kernel: <8003c720> from <8003ca78>
Oct 13 00:09:21 raspberrypi kernel: <8003ca78> from <80042754>
Oct 13 00:09:21 raspberrypi kernel: <80042754> from <8000fbc8>
```
Is it allright with SD card?

vpetersson, 2016-10-22:
Agreed. Looks like a faulty SD card. Try a fresh SD card and see if that helps.

closed 2016-10-22 by vpetersson

WTPLJim, 2016-10-24:
Card is brand new, 32 Gig Kingston.
Will the Pi support 64 (only spare I have here).

vpetersson, 2016-10-24:
The Pi is known to have issues with some bigger cards. I would think that 32GiB Kingston cards would work, but try with the 64GiB card to see if that makes a difference.

WTPLJim, 2016-10-25:
Haven't seen the hardware issue in logs since a reinstall.
Seeing the browser crash.
[piLogs.txt](https://github.com/wireload/screenly-ose/files/551324/piLogs.txt)

vpetersson, 2016-10-26:
Looks like something is wrong with git repository. Try the following:

```
$ cd ~/screenly
$ git reset --hard
$ git checkout production
$ git pull
```

WTPLJim, 2016-10-26:
Completed.
pi@raspberrypi:~ $ cd ~/screenly
pi@raspberrypi:~/screenly $ git reset --hard
HEAD is now at e332c69 Merge pull request #491 from ETexan/patch-1
pi@raspberrypi:~/screenly $ git checkout production
Switched to branch 'production'
Your branch is up-to-date with 'origin/production'.
pi@raspberrypi:~/screenly $ git pull
remote: Counting objects: 13, done.
remote: Compressing objects: 100% (5/5), done.
remote: Total 13 (delta 6), reused 5 (delta 5), pack-reused 3
Unpacking objects: 100% (13/13), done.
From https://github.com/wireload/screenly-ose
   e332c69..77c8d79  master     -> origin/master
Already up-to-date.

vpetersson, 2016-10-26:
Ok, that should make the error go away.