Screenly/Anthias #493: add support for HTTP Basic Authentication

TonyLHansen, 2016-10-09:
bottle.py fully supports HTTP Basic Authentication via `@basic_auth()` decorators. It just needs to be called within server.py.

Here is one implementation of it that has been tested and works. I tried doing a "git clone" so that I could do push you a pull request, but get a permission denied message; I'm obviously doing something wrong. Oh well. So instead, I'm enclosing context diffs for the needed changes.

Start with the configuration changes. I chose to do this using an [auth] configuration section and user and password parameters in screenly.conf.

The server.py changes consist of adding a single function in the model section, and calls to that function using the auth_basic() decorator calls. The decorator needs to be added to all methods that I felt need protecting, which is almost all of them.
  
Finally, we need the changes to settings.py to access the configuration file parameters. This consists of an addition to the default configuration, and a single new function to check the credentials.

[ansible.screenly.conf.diff.txt](https://github.com/wireload/screenly-ose/files/518580/ansible.screenly.conf.diff.txt)
[server.py.diff.txt](https://github.com/wireload/screenly-ose/files/518581/server.py.diff.txt)
[settings.py.diff.txt](https://github.com/wireload/screenly-ose/files/518582/settings.py.diff.txt)
[screenly.conf.diff.txt](https://github.com/wireload/screenly-ose/files/518579/screenly.conf.diff.txt)

vpetersson, 2016-10-10:
Thank you, @TonyLHansen. Could you please create a [Pull Request](https://help.github.com/articles/about-pull-requests/) for this so that we can do an audit and hopefully merge it into the master branch?

TonyLHansen, 2016-10-11:
As explained within the issue, I tried but got permission errors from github.

                Tony

vpetersson, 2016-10-12:
@TonyLHansen Here's what you need to do:

 * Fork the repository to your own Github account
 * Create a branch with the changes that fork
 * Initiate a pull request from that branch 

(You cannot push directly to this repo nor create a Pull Request directly in this repo.)

TonyLHansen, 2016-10-12:
see pull request in issue #495

closed 2016-10-12 by TonyLHansen