Screenly/Anthias #494: Server Error: Could not retrieve file. Check the asset URL.

cwonsey, 2016-10-12:
When trying to add a Web page that takes a while to load, I encounter this error.  I have several pages that worked great on the previous version, but that SD card became corrupt and I rebuilt the system with the latest version.  I have not been able to get any of these pages loaded after the change.  I have attempted to send headers more quickly in the transaction and still it gives the error in the subject if the entire page cannot be loaded a short duration.  I have loaded a very basic page to the same location and it is added without issue, this has to be a timeout value somewhere - I attempted to modify the timeout in server.py, but it made no difference.

If I add the asset while I have a simple page located at the URL, then upload the actual page to that URL, the page never loads.  I have tried to enable debug logging, but thre is no data in the /var/log/debug file after receiving this error.  I have attempted to locate screenly logs in TMP folders and have found no such logs.

cwonsey, 2016-10-12:
Just to update, I have no trouble accessing the page with CURL and Python request seems fine.
root@raspberrypi:/# python
Python 2.7.9 (default, Mar  8 2015, 00:52:26)
[GCC 4.9.2] on linux2
Type "help", "copyright", "credits" or "license" for more information.
>>> import requests
>>> r = requests.head('http://192.168.62.226/dailycalls/screen-dailydown.php')
>>> print r.status_code
200
>>> print r
<Response [200]>

vpetersson, 2016-10-12:
Hi @cwonsey. Could you please paste the output of `sudo journalctl -f`? The logs have been moved there from `/tmp` (with the migration to systemd).

cwonsey, 2016-10-13:
I have pasted the log below.  During this time I tried to add a URL that is displaying this error.  The page that it keeps skipping is a working page - to get it added, I had to upload a very basic page to this URL, add it to the system, then change the page back to my actual content.  With this page content, it will not allow me to add the page as an asset and pages that were added will not display, but there is nothing wrong with the page, it just takes up to 10 seconds to fully load the page.

-- Logs begin at Wed 2016-10-12 17:18:40 UTC. --
Oct 13 12:03:00 raspberrypi systemd[17395]: Reached target Default.
Oct 13 12:03:00 raspberrypi systemd[17395]: Startup finished in 17ms.
Oct 13 12:03:14 raspberrypi python[771]: Last update: 2016-10-12 15:55:44.342890
Oct 13 12:03:14 raspberrypi python[771]: get_next_asset
Oct 13 12:03:14 raspberrypi python[771]: refresh_playlist
Oct 13 12:03:14 raspberrypi python[771]: refresh: counter: (0) deadline (2016-10                                                                                                                                                             -19 16:41:00) timecur (2016-10-13 12:03:14.761534)
Oct 13 12:03:14 raspberrypi python[771]: get_next_asset after refresh
Oct 13 12:03:14 raspberrypi python[771]: get_next_asset counter 0 returning asse                                                                                                                                                             t 1 of 2
Oct 13 12:03:22 raspberrypi sudo[17420]: pi : TTY=pts/1 ; PWD=/home/pi ; USER=ro                                                                                                                                                             ot ; COMMAND=/bin/journalctl -f
Oct 13 12:03:22 raspberrypi sudo[17420]: pam_unix(sudo:session): session opened                                                                                                                                                              for user root by pi(uid=0)
Oct 13 12:03:24 raspberrypi python[771]: Asset DailyUp at http://192.168.62.226/                                                                                                                                                             dailycalls/screen-dailyup.php is not available, skipping.
Oct 13 12:03:25 raspberrypi python[771]: Last update: 2016-10-12 15:55:44.342890
Oct 13 12:03:25 raspberrypi python[771]: get_next_asset
Oct 13 12:03:25 raspberrypi python[771]: refresh_playlist
Oct 13 12:03:25 raspberrypi python[771]: refresh: counter: (0) deadline (2016-10                                                                                                                                                             -19 16:41:00) timecur (2016-10-13 12:03:25.315247)
Oct 13 12:03:25 raspberrypi python[771]: get_next_asset after refresh
Oct 13 12:03:25 raspberrypi python[771]: get_next_asset counter 0 returning asse                                                                                                                                                             t 2 of 2
Oct 13 12:03:25 raspberrypi python[771]: Showing asset Test (webpage)
Oct 13 12:03:25 raspberrypi python[771]: Asset URI http://192.168.62.234
Oct 13 12:03:25 raspberrypi python[771]: Already showing http://192.168.62.234,                                                                                                                                                              reloading it.
Oct 13 12:03:25 raspberrypi python[771]: Sleeping for 15
Oct 13 12:03:40 raspberrypi python[771]: Last update: 2016-10-12 15:55:44.342890
Oct 13 12:03:40 raspberrypi python[771]: get_next_asset
Oct 13 12:03:40 raspberrypi python[771]: refresh_playlist
Oct 13 12:03:40 raspberrypi python[771]: refresh: counter: (0) deadline (2016-10-19 16:41:00) timecur (2016-10-13 12:03:40.827670)
Oct 13 12:03:40 raspberrypi python[771]: get_next_asset after refresh
Oct 13 12:03:40 raspberrypi python[771]: get_next_asset counter 0 returning asset 1 of 2
Oct 13 12:03:50 raspberrypi python[771]: Asset DailyUp at http://192.168.62.226/dailycalls/screen-dailyup.php is not available, skipping.
Oct 13 12:03:51 raspberrypi python[771]: Last update: 2016-10-12 15:55:44.342890
Oct 13 12:03:51 raspberrypi python[771]: get_next_asset
Oct 13 12:03:51 raspberrypi python[771]: refresh_playlist
Oct 13 12:03:51 raspberrypi python[771]: refresh: counter: (0) deadline (2016-10-19 16:41:00) timecur (2016-10-13 12:03:51.384970)
Oct 13 12:03:51 raspberrypi python[771]: get_next_asset after refresh
Oct 13 12:03:51 raspberrypi python[771]: get_next_asset counter 0 returning asset 2 of 2
Oct 13 12:03:51 raspberrypi python[771]: Showing asset Test (webpage)
Oct 13 12:03:51 raspberrypi python[771]: Asset URI http://192.168.62.234
Oct 13 12:03:51 raspberrypi python[771]: Already showing http://192.168.62.234, reloading it.
Oct 13 12:03:51 raspberrypi python[771]: Sleeping for 15
Oct 13 12:04:06 raspberrypi python[771]: Last update: 2016-10-12 15:55:44.342890
Oct 13 12:04:06 raspberrypi python[771]: get_next_asset
Oct 13 12:04:06 raspberrypi python[771]: refresh_playlist
Oct 13 12:04:06 raspberrypi python[771]: refresh: counter: (0) deadline (2016-10-19 16:41:00) timecur (2016-10-13 12:04:06.916594)
Oct 13 12:04:06 raspberrypi python[771]: updating playlist due to database modification
Oct 13 12:04:06 raspberrypi python[771]: update_playlist
[... cut ...]
Oct 13 12:05:09 raspberrypi python[771]: refresh: counter: (0) deadline (2016-10-19 16:41:00) timecur (2016-10-13 12:05:09.597671)
Oct 13 12:05:09 raspberrypi python[771]: get_next_asset after refresh
Oct 13 12:05:09 raspberrypi python[771]: get_next_asset counter 0 returning asset 2 of 2
Oct 13 12:05:10 raspberrypi python[771]: Showing asset Test (webpage)
Oct 13 12:05:10 raspberrypi python[771]: Asset URI http://192.168.62.234
Oct 13 12:05:10 raspberrypi python[771]: Already showing http://192.168.62.234, reloading it.
Oct 13 12:05:10 raspberrypi python[771]: Sleeping for 15

vpetersson, 2016-10-13:
Can you try the following *from* the Pi?

```
$ python
>>> import requests
>>> r = requests.head("http://192.168.62.226/ dailycalls/screen-dailyup.php")
>>> print r.status_code
```

cwonsey, 2016-10-13:
I posted that in my second comment, I received a 200 OK with that test.  I also tested with CURL and was able to load the page on the PI.

vpetersson, 2016-11-30:
@over64 any ideas?

firepro20, 2017-05-17:
I have had the same problem two days ago. I found out that since it is internal, the Pi has no entry to the contents location. What I did was add an entry in my hosts file to map the domain with an IP.

 `sudo nano /etc/hosts`

and add the entry like so

`192.168.0.xxx     mydomain.com`

vpetersson, 2017-05-18:
@firepro20 Oh yeah, if you don't have an internal DNS that can do those lookups, you need to add it to `/etc/hosts`.

vpetersson, 2017-07-02:
@cwonsey Are you able to reproduce this on the latest development version? Please comment if this is still an issue and we'll re-open the issue.

closed 2017-07-02 by vpetersson