Screenly/Anthias #504: Add random on asset with a level on % of display

forum6691, 2016-11-01:
Would it be possible to add this function:
When I set shuffle display for the configuration of  the assets, they are displayed in random. That's fine, but I 'd like some asset to appear more or less than other.
It could be achieve in adding a percentage parameter for the asset.
If the % is 100 %, the asset appears normaly.
If the % is 50 % the asset appears twice less than other asset put at 100 %
if the % is 50 % the asset appears four times less than other asset put at 100 % and so on.

I think it's an interesting function for screenly.

vpetersson, 2016-11-01:
I'm not sure how that would work, but if you could write out that in a simple proof-of-concept in a pull request, we could look into it.

forum6691, 2016-11-01:
I'm not sure to understand well you Victor.
I've created a pull request with the same information that this issue. is it ok ?

Why this new function ? When you do advertising on a screen, sometimes, you want to insist on a or some asset with are important at the moment, so these asset must appear more than the other. 
This % parameter could be used for that.
You must divide the percentage by 100, then take the inverse of this value and you get the multiplicator (ex 25 % give you 4, 12.5 % give you 8,...) for the frequency of occurence.

vpetersson, 2016-11-01:
No, we need a pull request with the actual code that actually does this, not a just a verbal description of it.

forum6691, 2016-11-01:
I'm not a specialist of this code, so it would be difficult for me to add the correct code on the correct source code

vpetersson, 2016-11-01:
Ok. Leaving this as a feature request then.

edav8458, 2017-06-15:
A feature like this would be useful to us too, though I think the ideal implementation for us would be the grouping/linking of assets, then selecting how often they were shown in terms of "show every x assets. 

E.g - say I want to show a news website, then show a timetable, then flick through some random images before showing the news and timetable again:

In the asset list, I'd make a "group" that contains the news website and the timetable, which then gives me the option to "show group contents every X assets." It would probably make sense that any groups were pinned to the top of the asset list, and the contents could be hidden or shown to make things cleaner.

Sadly I'm not much of a coder so can't offer any sample code, but I think this would be much cleaner and simpler than weighting the probability of each asset being shown.

closed 2018-10-19 by stale[bot]