Screenly/Anthias #506: Option to let the displayed Webpage scroll automatically

nilsbyte, 2016-11-04:
To see more than just the top of a page, an option to let the displayed webpage scroll automatically would be cool. This feature would consist of one on/off toggle and a textbox with the value in seconds how long the scrolling should take to reach the bottom of the page. 

Is this possible?

vpetersson, 2016-11-04:
Interesting idea. Not sure how easily this is attainable.

firepro20, 2017-05-17:
I think this has to do with the page itself. If you have control over the page, then use CSS to obtain the desired result.

nilsbyte, 2017-05-17:
In most cases you won't have control over the sites. So it would be better to have the option inside Screenly.

vpetersson, 2017-05-18:
To be honest, I don't think this is a feature that is desirable to implement. It might be possible to do in theory, but I don't think the user experiment would be very good.

closed 2017-05-18 by vpetersson

securitygeneration, 2020-01-16:
I'm surprised this was closed and the feature disregarded. This is literally one of the main features I'm looking for in a dashboard solution. I currently do this using Chrome and the simply scrolling extension - and it simply scrolls down the page at a slow pace, but allows the user to see all of the data on it. This is essential when trying to display information from a dashboard that doesn't fit onto a single screen's worth of space. Fundamental imo.

vpetersson, 2020-01-17:
IMHO this is something that should be fixed in the dashboard. It should render to 1920x1080 (like most proper dashboard software does). Most tooling (such as Grafana) allows you to create multiple views (each optimized for the playback resolution). The players job is to toggle through a set of assets that renders properly.