Screenly/Anthias #509: Problems with animated gif

forum6691, 2016-11-06:
I try to display simple animated gif, but some pictures block the screenly. 
They just begin to show the animation and block all the following slides-show.

Is a known problem ? Is there a solution ?

vpetersson, 2016-11-16:
I think it should work, but it depends on how you are viewing them. Are you adding it as an asset or is it part of a website?

forum6691, 2016-11-19:
I add  it as an asset

vpetersson, 2016-11-20:
Alright, yeah we might need to refactor the image playback code to make this work. The issue (i think) is that we are displaying images as background images to piggy back on some automatic resizing elements etc.. It's very likely that this is what prevents GIFs to be rendered properly.

forum6691, 2016-11-20:
OK thanks victor

closed 2016-11-20 by forum6691