Screenly/Anthias #515: Add support for assets with basic auth

vpetersson, 2016-11-23:
Contrary to what we previously thought, UZBL does indeed support Basic Auth. It's just supported through a somewhat odd format where you need to escape the `@`  (i.e. `http://username:password\@domain:port/path`. Kudos to @sodabrew for finding out about this. More information about this can be found [here](https://github.com/uzbl/uzbl/issues/311#issuecomment-249572291).

In order to add this support, we need to add custom handling for this. 

* In the HEAD check (i.e. python request), we might need to parse the credentials and pass them on as follows: `requests.head('https://domain/path', auth=HTTPBasicAuth('username', 'password')`
* In the UZBL call, we need to modify the `uri` we pass on to UZBL as described above (i.e. `http://username:password\@domain:port/path`).

over64, 2016-12-06:
Where should we get username and password?

vpetersson, 2016-12-06:
We'd parse it from the URL string, just like most browsers do.

I.e., you'd add an asset with the URL `http://username:password@domain:port` and we'd parse it from there.

vpetersson, 2016-12-13:
@over64 I've reopened this issue. @sodabrew raised a valid point in #526.

closed 2016-12-20 by vpetersson

farhangithub27, 2019-02-07:
> We'd parse it from the URL string, just like most browsers do.
> 
> I.e., you'd add an asset with the URL `http://username:password@domain:port` and we'd parse it from there.

Hi @vpetersson , I'm trying to configure web asset with URL  format as `http://username:password@domain:port` 
following cases I have used example. I'm trying to land at the home page of my reporting server.

- http://myusername:mypassword@myhost.mydomain:myport/reportingserver/flow.html?_flowid=searchFlow
This url is not getting accepted by screenly. Throws out error message 
 **Server Error: Could not retrieve file. Check the asset URL**.I guess uzbl doesn't accept _flowid parameter?
- http://myusername:mypassword@myhost.mydomain:myport/reportingserver/flow.html
This one doesn't get accepted by screenly throws out same error message as above.
- http://myusername:mypassword@myhost.mydomain:myport/reportingserver/login.html
This one gets accepted but nothing gets displayed.
- http://myhost.mydomain:myport/reportingserver/login.html
This url gets accepted and login screen gets displayed but then again I cannot use keyboard.

FYI a get request in POSTMAN and firefox  browser works fine with standard get request that lands me to the home page of the server. Postman request below is
http://myhost.mydomain:port/reportingserver/flow.html?_flowid=searchFlow&username=myusername&password=mypassword
Also
I have installed uzbl browser on my development system and used URL used for postman
I got logged in successfully through uzbl but it didn't recognize _flowid parameter. So it logged me in but didn't display the home page. I just used mouse and menu bar of the presented page to get to home page.
**This means uzbl takes in username and password parameters fine in get request while _flowid was ignored.**
So I tried to use this request format in screenly webasset configuration but it throws out same error message:
**Server Error: Could not retrieve file. Check the asset URL**

Any suggestions how to use your suggested format or anyother to login into my reporting server through screenly web asset.