Screenly/Anthias #516: the etc/network/interfaces overwrites itself by reboot

Fabi2332, 2016-12-02:
Hey folks,

Tried to enabe Wifi following this guide :
http://www.howtogeek.com/167425/how-to-setup-wi-fi-on-your-raspberry-pi-via-the-command-line/

After restarting the interface with `sudo ifdown wlan0` and `sudo ifup wlan0` everything worked fine.
But just after the next reboot, the /etc/network/interfaces changes to default.
I'm using the Pi 3 and its build in wifi chip. 

Don't know if this is a bug or just me. Sorry for not beeing that good in using screenly on Pi ;) 
Thanks a lot for any help givin :)

vpetersson, 2016-12-02:
Yes, we should have communicated this more clearly, but you need to use [these instructions](https://support.screenly.io/hc/en-us/articles/212107126-Can-I-use-a-WiFi-adapter-with-Screenly-Pro-). Alternatively, you can disable the service called Screenly Net Mgr.

closed 2016-12-02 by vpetersson

thhartung, 2016-12-11:
I followed the instructions from your link above, still no wifi.

> Create a file called network.ini on the boot partition (or just copy the existing sample file)

There alredy was an network.ini but no sample file. the present network.ini was already full with a comfiguration, but it didnt look like your example, insstead of your clean formated example it was a list seperated by comma.
Which version is the rigt one?

following your guide my network.ini now looks like this:

[wlan0]
ip=192.168.XXX.XXX
netmask=255.255.255.0
gateway=192.168.XXX.XXX
ssid=my.network
passphrase=mypassphrase
ntp=0.pool.ntp.org,1.pool.ntp.org

Still no coneection, etc/network/interfaces remains in sample state

vpetersson, 2016-12-11:
First, are you running the latest updates in from master (i.e. the development branch). If not, start there.

If you are, then please provide the output of the following:

```
$ cat /boot/network.ini
$ ls -lah /boot
$ cat /etc/network/interfaces
```

julian-labuschagne, 2017-09-07:
How do one disable Screenly Net Mgr I have exactly the same issues.

I don't have a /boot/network.ini file

And my /etc/network/interfaces keeps being overwritten after boot.

With a line at the the top

> # Generated by Screenly Network Manager

Where I change the template Screenly uses to generate the interfaces file?

vpetersson, 2017-09-07:
@julian-labuschagne you need to disable the systemd service.

dmekersa, 2018-03-06:
I resolved by editing /boot/interfaces