Screenly/Anthias #522: network.ini not applied at start up

lamrod, 2016-12-09:
As part of the static network configuration a network.ini configuration file must be placed in the /boot partition.
During the boot process this file is supposed to be parsed and the configurations made.

This is not happening

vpetersson, 2016-12-09:
Strange. I've accepted your PR. Does that solve the issue? Thanks for that btw.

lamrod, 2016-12-09:
It has worked on the 3 devices we have running in our office.

vpetersson, 2016-12-09:
Cool. Closing out. Thanks.

closed 2016-12-09 by vpetersson

lamrod, 2016-12-09:
I've just noticed that the Ansible role doesn't actually reference the file that I edited in my PR. Is this intended or have I just missed something?

vpetersson, 2016-12-09:
Ah yes, that file is actually not in use. That's from the old system. It was replaced by [this](https://github.com/wireload/screenly-ose/blob/master/ansible/roles/network/files/screenly-net-manager.service) file. Let me delete this file to avoid confusion.

vpetersson, 2016-12-09:
Resolved in #524. Just waiting for a LGTM from @over64 before merging.