Screenly/Anthias #527: Improved web loading - Replace UZBL with Chromium

vpetersson, 2016-12-13:
As discussed in #257, it would be desirable to replace UZBL with a more modern browser. Chromium appears to be the best out-of-the-box browser for the Raspberry Pi right now. 

The issue however is that it's not as easy to remote control as UZBL, but it's worth exploring.

The primary way to remote control Chromium/Chrome appears to be using the [JavaScript API](https://developer.chrome.com/extensions/api_index). There's also a project called [Chromix-too](https://github.com/smblott-github/chromix-too) which provides a command line tool on top of the JS API.

As a side-note, my initial idea was to use Selenium, but unfortunately [this is not an option](https://stackoverflow.com/questions/31190164/using-selenium-on-raspberry-pi-with-chromium) due to the lack of ARM support in the driver.

gustavosbarreto, 2017-02-01:
Hi @vpetersson 

Do you know the ```electric-webview```? It's a scriptable WebView built on top of QtWebEngine which provides a simple protocol over Unix Socket, TCP or WebSocket to communicate with its WebView.

I'm the author of ```electric-webview```. Please let me if you want to evaluate.

https://github.com/gustavosbarreto/electric-webview

vpetersson, 2017-02-01:
@gustavosbarreto Interesting idea. Have you tested it on the Raspberry Pi? We have been working with QT on the Raspberry Pi in for another project and from that experience, we know that WebKit etc on the Raspberry Pi requires a fair amount of patches to work well.

gustavosbarreto, 2017-02-07:
@vpetersson Yes, I have tested it on the Raspberry Pi. Works fine. I'm using Yocto to build a image for Raspberry Pi 3.

[image]

vpetersson, 2017-02-08:
@gustavosbarreto Very cool. Do you have an apt package that we can test with?

gustavosbarreto, 2017-02-08:
@vpetersson I don't have a DEB package, sorry, but you can build it yourself following the instructions at https://github.com/gustavosbarreto/electric-webview/blob/master/README.md#building

vpetersson, 2017-02-10:
Cool. Looks straight forward. We will investigate this going forward.

xlegs, 2017-03-13:
Has anyone managed to make a build with this? How would one go about replacing UZBL, anyway?

vpetersson, 2017-03-13:
@xlegs it is a fair amount of work. You'd have to modify all signaling done in `viewer.py` as well as all the initiation process for the viewer.

vpetersson, 2017-11-21:
Another alternative:
https://github.com/resin-io-playground/resin-wpe

antonmolodykh, 2017-11-23:
| Project    | Website | Hardware acceleration | Updated in the last month | Scriptable |
| --- | --- | --- | --- | --- |
| Luakit | https://github.com/luakit/luakit | No | Yes | No |
| Chromix-too | https://github.com/smblott-github/chromix-too | ? | No | Yes |
| Electric-webview | https://github.com/gustavosbarreto/electric-webview | No | No | Yes |
| Kweb | https://www.raspberrypi.org/forums/viewtopic.php?t=40860 | Yes | No | No|

I will add to the list. I will be glad to the new options

vpetersson, 2018-06-22:
Just an update to anyone following this issue: you can now try the new browser out by running the installer and selecting the 'Experimental' version.

vpetersson, 2018-06-22:
The new browser scores 253 with [HTML5test](https://html5test.com/).

vpetersson, 2018-10-02:
Closing out this ticket as it is work in progress. The new browser is available in the experimental branch and the code can be found [here](https://github.com/Screenly/screenly-ose-webview).

closed 2018-10-02 by vpetersson

ansonhoyt, 2018-10-18:
@vpetersson since this is ongoing, is there an open ticket tracking it for interested folks to follow your progress?

vpetersson, 2018-10-18:
@ansonhoyt There's already a version for this. You can test it out using the experimental branch. We don't have any specific issue tracking for it as they will just go in as normal issues.

pjsignage, 2019-02-01:
@vpetersson Is the experimental version available to deploy using balena.io?

vpetersson, 2019-02-21:
@pjsignage I think it should, yes. I haven't tested it myself.

lztrade, 2019-06-18:
@vpetersson Could you please re-evaluate the current possibility of using Chromium/Selenium, there seems to be an ARM driver mentioned in this conversation:  [https://stackoverflow.com/questions/49796818/webdriverexception-message-service-usr-lib-chromium-browser-chromedriver-unex](url)
I mean this driver:    https://github.com/electron/electron/releases/download/v1.6.0/chromedriver-v2.21-linux-armv7l.zip
There is also a driver mentioned here: [https://stackoverflow.com/questions/38732822/compile-chromedriver-on-arm](url)
Is this what is needed for using Chromium as a viewer?  Thank you.

vpetersson, 2019-06-18:
@lztrade we've already invested significant effort in the browser rewrite. This is already available in the  experimental branch. Hence we would need a *very* good reason to throw this out.