Screenly/Anthias #529: InsecureRequestWarning when displaying SSL secured sites over HTTPS

nilsbyte, 2016-12-14:
SSL secured sites are displayed but the log says

´´´
screenly python[741]: /usr/local/lib/python2.7/dist-packages/requests/packages/urllib3/connectionpool.py:821: InsecureRequestWarning: Unverified HTTPS request is being made. Adding certificate verification is strongly advised. See: https://urllib3.readthedocs.org/en/latest/security.html
´´´

I followed the steps at https://urllib3.readthedocs.org/en/latest/security.html but I think you have to change something in the python scripts of Screenly.

vpetersson, 2016-12-14:
Yeah this is a known issue. Doesn't really cause any issues tho.

leonelwilliams, 2018-01-08:
One option could be to verify things by default but add a per-asset checkbox to insecurely disable certificate verification if required. (One that I'd keep unchecked but I understand that not everyone has the luxury unfortunately.)

leonelwilliams, 2018-08-13:
I don't have use for screenly anymore at the moment, unfortunately, but loading resources with no certificate verification isn't really the best practise as there's no guarantee the request or response wasn't viewed or modified.

vpetersson, 2018-08-13:
@leonelwilliams Agreed. This was partially due to the fact that we had to support older versions of Raspbian. We are now moving to docker, which should help us keep things more up to date.

closed 2018-08-13 by vpetersson