Screenly/Anthias #531: DNS not work after reboot

scyhe, 2016-12-22:
After reboot screenly dns servers from resolv.conf are deleted. I have line : dns-nameservers 8.8.8.8 in /etc/network/interfaces and line dns=8.8.8.8,8.8.4.4 in /boot/network.ini, but everytime when reboot RaspberryPI DNS not working, file resolv.conf is empty.  Version OSE : 2016-12-01

vpetersson, 2016-12-22:
Hmm it shouldn't be empty. However if you're using a custom DNS, you need to follow [these](https://support.screenly.io/hc/en-us/articles/212107026-Can-I-use-a-custom-DNS-as-well-as-DHCP-with-Screenly-) instructions or simply disable the Screenly network configuration tool.

scyhe, 2016-12-22:
I have this lines in /boot/network.ini 👍 
```
[eth0]
mode=static
ip=10.0.0.100
netmask=255.255.255.0
gateway=10.0.0.1

[generic]
dns=8.8.8.8,8.8.4.4
```

vpetersson, 2016-12-22:
Can you cat `/etc/network/interfaces` just after a reboot?

scyhe, 2016-12-22:
```

pi@raspberrypi:~ $ cat /etc/network/interfaces
# Generated by Screenly Network Manager
auto lo
  iface lo inet loopback

auto eth0
  iface eth0 inet static
  address 10.20.0.199
  netmask 255.255.255.0
  gateway 10.20.0.254
  dns-nameservers 8.8.8.8

```

vpetersson, 2016-12-22:
The interfaces file does look correct. Strange.

vpetersson, 2017-04-11:
Ahhh. I see what's going on here. There were changes upstream that switched to using `resolvconf
`, so the changes that `screenly_net_mgr.py` are being overwritten. Will need to investigate how to fix this best.

vpetersson, 2017-04-11:
Fixed in #557. Sorry for the delay sorting this one out.

closed 2017-04-13 by vpetersson