Screenly/Anthias #532: Hidden Wireless

rsenio, 2016-12-28:
Using these directions don't seem to work.
https://support.screenly.io/hc/en-us/articles/218364983-Can-I-connect-to-a-hidden-SSID-WiFi-Access-Point-?utm_source=ScreenlyApp.com&utm_campaign=112e6e4592-srly_newsletter_2016-12_general&utm_medium=email&utm_term=0_5c47388076-112e6e4592-439022077

Once the changes are made to the network.ini file, and the Pi rebooted. I'm not seeing any reference to the hidden ssid in the /etc/network/interfaces file like I am with the other information populated in network.ini. Therefor wifi isn't working...again unless I revert back to the normal way of using the interfaces file

For example:

Network.ini contains
[wlan0]
ssid=SSIDName
passphrase=wifi password
hidden_ssid=true

After a reboot the /etc/network/interfaces file contains
# Generated by Screenly Network Manager
auto lo
  iface lo inet loopback

auto wlan0
  iface wlan0 inet dhcp
  wireless-power off
  wpa-ssid "SSIDName"
  wpa-psk "wifi password"

vpetersson, 2016-12-30:
@rsenio Are you using the master/developer version? This is not available in the stable release yet.

rsenio, 2016-12-30:
Oh ok, I was only going by what the newsletter email had stated. I didn't realize it was still in the beta. I'll wait for an update.

vpetersson, 2016-12-30:
The announcements in the newsletter are primarily for Pro. Some of the changes (such as this) makes it to OSE.

vpetersson, 2017-04-13:
This should work now. Try upgrading to the latest development version. If you're still having issues, please let us know.

Also, you might want to take a look at our new [Screenly Configuration Wizard](https://wizard.screenly.io/).

closed 2017-04-13 by vpetersson