Screenly/Anthias #534: No open port after update

CueballUK, 2016-12-30:
Hi, I have done an update on Screenly and since the update I have lost the webserver.
A check using netstat shows only port 22 and port 6000 open.

vpetersson, 2016-12-30:
Hard to say without any further logs. Please do the following:

 * Run the update again
 * Provide us with the output from `sudo journalctl`

CueballUK, 2016-12-30:
Hi I have found this in my logs.
`Dec 30 21:23:33 screenly-ebt systemd[1]: screenly-web.service holdoff time over, scheduling restart.
Dec 30 21:23:33 screenly-ebt systemd[1]: Unit screenly-web.service entered failed state.
Dec 30 21:23:44 screenly-ebt python[13593]: Bottle v0.12.9 server starting up (using GunicornServer(threads=2, timeout=20))...
Dec 30 21:23:44 screenly-ebt python[13593]: Listening on http://0.0.0.0:8080/
Dec 30 21:23:44 screenly-ebt python[13593]: Hit Ctrl-C to quit.
Dec 30 21:23:45 screenly-ebt python[13593]: Error: class uri 'gunicorn.workers.gthread.ThreadWorker' invalid or not found:
Dec 30 21:23:45 screenly-ebt python[13593]: [Traceback (most recent call last):
Dec 30 21:23:45 screenly-ebt python[13593]: File "/usr/local/lib/python2.7/dist-packages/gunicorn/util.py", line 142, in load_class
Dec 30 21:23:45 screenly-ebt python[13593]: mod = import_module('.'.join(components))
Dec 30 21:23:45 screenly-ebt python[13593]: File "/usr/lib/python2.7/importlib/__init__.py", line 37, in import_module
Dec 30 21:23:45 screenly-ebt python[13593]: __import__(name)
Dec 30 21:23:45 screenly-ebt python[13593]: File "/usr/local/lib/python2.7/dist-packages/gunicorn/workers/gthread.py", line 37, in <module>
Dec 30 21:23:45 screenly-ebt python[13593]: """)
Dec 30 21:23:45 screenly-ebt python[13593]: RuntimeError:
Dec 30 21:23:45 screenly-ebt python[13593]: You need to install the 'futures' package to use this worker with this
Dec 30 21:23:45 screenly-ebt python[13593]: Python version.
Dec 30 21:23:45 screenly-ebt python[13593]: ]
Dec 30 21:23:45 screenly-ebt systemd[1]: screenly-web.service: main process exited, code=exited, status=1`

vpetersson, 2016-12-30:
Hmm. Can you post the full result from:

```
bash <(curl -sL https://www.screenly.io/install-ose.sh)
```

Also, please answer yes to all questions except for the development branch one.

CueballUK, 2016-12-30:
Seems I don't have journalctl on my system. Even though I do have systemd!

CueballUK, 2016-12-30:
These are old installs and are still on wheezy. I'm going to upgrade to jessie and get back.

vpetersson, 2016-12-30:
In theory, it should still work on Wheezy, but yeah, I'd recommend that you switch to the latest disk image that is based on Jessie.

closed 2016-12-30 by vpetersson